firmata/arduino #183: compile firmata and upload from command line

legrostdg, 2015-01-10:
Is there a way to compile StandardFirmata.ino and to upload it to a board without using the IDE?

I tried to use [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile) but I got https://github.com/sudar/Arduino-Makefile/issues/117

soundanalogous, 2015-01-10:
There used to be but I removed the makefiles a while back since no one was keeping them up to date. Try this: https://github.com/firmata/arduino/blob/e2ae488888a005fbaec1ff302efce7140a8f3ba2/examples/StandardFirmata/Makefile. If it works I can add it back.

ajfisher, 2015-01-11:
I use ino for this pretty much all the time. Involves a little bit of
symlinking things to the right place but vim + ino is a great dev combo.

soundanalogous, 2015-01-11:
@ajfisher do you need the makefile to use ino or does it generate its own?

ajfisher, 2015-01-11:
It does it's own though you can give it one to do things like set device
ports, arduino types etc.

soundanalogous, 2015-10-18:
I'll leave this open in case anyone wants to contribute via creating documentation of how to do this and/or contributing any configs, makefiles, etc to make this happen.

ajfisher, 2015-10-18:
I'll put some notes together around this as I've now got rid of using INO due to the project going stale. 

With Interchange I've moved across to using the Arduino CLI tool and it works very well. Will put some documentation together relating to that process for those who are looking to do this CLI only.

soundanalogous, 2016-05-22:
Since this issue is not specific to Firmata and is really a broader issue about compiling and uploading sketches via the command line, I'm closing this issue.

closed 2016-05-22 by soundanalogous

carrete, 2024-03-15:
I understand this is a very old issue, but this did show-up in my search for an answer to the same question. With [arduino-cli](https://arduino.github.io/arduino-cli/0.35/), these steps worked for me:

    arduino-cli lib install firmata servo
    arduino-cli compile -b arduino:avr:uno .
    arduino-cli upload -b arduino:avr:uno -p /dev/ttyACM0 .
    
This was with a sketch that I had previously created in the IDE.