firmata/arduino #185: 2.4 spurious data

MrYsLab, 2015-02-07:
2.3 does not exhibit this problem. PyMata is receiving a spurious stream of unsolicited data at startup that is crashing the client. No commands have been sent to Firmata. Data is received as single bytes and is formatted here with the word data followed by b and then the byte data.

data b'\xe0'
data b'S'
data b'\x02'
data b'\xe1'
data b'9'
data b'\x02'
data b'\xe2'
data b'\x00'
data b'\x00'
data b'\xe3'
data b'\x05'
data b'\x04'
data b'\xe4'
data b'j'
data b'\x03'
data b'\xe5'
data b'd'
data b'\x02'
data b'\xe6'
data b'\x00'
data b'\x00'
data b'\xe7'
data b'\x00'
data b'\x00'
data b'\xe8'
data b'\x00'
data b'\x00'
data b'\xe9'
data b'\x00'
data b'\x00'
data b'\xea'
data b'\x00'
data b'\x00'
data b'\xeb'
data b'\x00'
data b'\x00'

soundanalogous, 2015-02-07:
Nothing has changed in Firmata.cpp since 2.3.6 that would impact startup code. Did you happen to load StandardFirmataYun rather than StandardFirmata by chance or are you using your own skech?

soundanalogous, 2015-02-07:
It actually looks like noise is being picked up on your analog pins. e0 (ANALOG_MESSAGE) is analog pin 1, e1 is analog pin 2, etc. Do you have analog pin reporting enabled by default for all analog pins? That could produce noise on any analog pins that are not attached to a sensor or other device. It's difficult for me to diagnose what the cause is without more info.

soundanalogous, 2015-02-07:
Figured it out. Here's the root of the issue you're experiencing: https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L352-L355. This is new as of Firmata 2.4.

The problem is that when all analog pins are set to ANALOG input in the `systemResetCallback` (which is called at the end of the `setup` function), the analog pin is read an reported. This will essentially report noise for any unconnected analog pin. I'll need to suppress [this line](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L352-L355) from being called when `systemResetCallback` is called by setting a flag before the default analog input is set and clearing it after. Then check that flag when reportAnalogCallback is called with a value parameter of 1.

MrYsLab, 2015-02-07:
Great work thanks!

soundanalogous, 2015-02-07:
This issue also affects ConfigurableFirmata but a fix for this issue in the configurable branch is not so simple because of the way features are split among separate classes.

soundanalogous, 2015-02-07:
@MrYsLab This be fixed now for StandardFirmata and StandardFirmataYun. Please sync with the master branch and let me know if you still encounter the issue. I'll cut a new release once verified.

MrYsLab, 2015-02-07:
@soundanalogous Works like a champ! Thanks for fixing this so quickly.

closed 2015-02-07 by MrYsLab