firmata/arduino #194: Firmata over Mqtt

mcinnes01, 2015-04-12:
Hi,

I have been playing around with firmata for a short time and think its fantastic. My main aim is to use it as an OS for controlling the arduino (ideally using node-red), but to have my arduinos independent and using mqtt. I've played with EtherPort etc and thats great, but I really like the idea of using mqtt to communicate. That what I can use node-red-contrib-gpio to communicate directly to the arduino over mqtt. Do you think this would be possible and secondly can you give me some pointers of where to begin if I wanted to do this? I guess I need to replace the serial comms bit right?

Many thanks

Andy

soundanalogous, 2015-04-12:
I don't know anything about Mqtt. You should ask this question on the Firmata gitter: https://gitter.im/firmata/arduino?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge. I think there are some people on that list who are familiar with Mqtt. I have no plans for adding Mqtt support myself so you'll have to rally some contributors.

jacobrosenthal, 2015-04-12:
I did this for Octoblu's proprietary (though open sourced) mqtt
implementation
https://github.com/octoblu/microblu_mqtt

Firmata writes to a Stream buffer which b64 encodes incoming data and
writes it to Nick Ollearys mqtt library.

mcinnes01, 2015-04-12:
Hi @jacobrosenthal 

Cool thanks, do you think it would be possible for me to use microblu_mqtt and use it with node-red-contrib-gpio, I know you use UUID and tokens but I guess these can just be seen as a user name and password for connecting to mqtt right? Do you think there would be much I would need to change to get it up and running?

Many thanks

Andy

monteslu, 2015-04-12:
node-red-contrib-gpio uses https://github.com/monteslu/mqtt-serial which assumes the data coming across is binary and node base64 encoded. 

UUID/token directly translates to username/password so there shouldnt be any problems there.

Will definitely need a new sketch, but shouldn't be much different than the microblu_mqtt one besides using binary instead of base64 encoding.  I'd help with this but I'm not the greatest at C, and I don't have an ethernet/wifi shield handy.  I know at least a couple of node-red users that would appreciate the updated sketch though :)

mcinnes01, 2015-04-12:
Hi @monteslu

Great thanks for the info :) I'm no C expert either, .net by day :-/

Any way I will give it a go and see what I can get out, I guess a good start would be getting something to publish out on mqtt, am I right in assuming the UUID is the topic as well as the user name?

Thanks again

Andy

monteslu, 2015-04-12:
The send and receive topics can be whatever you want as long as they're different.

You might want to try the firmata over mqtt pattern without a shield on standard firmata first to make sure your mqtt setup is good. I like using mosca for a simple test server.

Take a look t the mqtt-serial examples folder for running a mqtt/firmata client on node that attaches to another mqtt node client that is relaying to a serial connected arduino.

mcinnes01, 2015-04-14:
Hi @monteslu 

I have started trying to get the node red node to connect to the updated microblu sketch I have made, I can see the arduino connecting to my mqtt server and even get it to publish out an "I'm alive" message. However in node red, the nodebot with mqtt option selected just says connecting and I can see no messages at all going out on the mqtt topic. 

I have use the publish and subscribe topics from my arduino sketch just in reverse, so that the nodered node will publish to the topic the arduino is subscribed to and vice versa. I'm watching both topics in MQTTlens.

Do you have any ideas of what I should expect to see from node red, and am I correct in expecting to see it published to the mqtt topic?

Many thanks

Andy

mcinnes01, 2015-04-16:
Hi @jacobrosenthal 

I wonder if you can point me in the right direction in terms of not base 64 encoding but just sending/receiving binary over mqtt for firmata, as commented above that node-red-contrib-gpio does. I am currently able to connect my arduino to my mqtt server, but I can't see anything being published to an mqtt topic from node-red-contrib-gpio at all and would expect something for the connection to be established between the arduino and node-red?

Any pointers would be greatly appreciated :)

Andy

jacobrosenthal, 2015-04-16:
@mcinnes01 B64 is implemented in  microblu, copy whats there. I recommend liberal Serial.printlin to see what the arduino thinks it sending.

mcinnes01, 2015-04-18:
Hi @monteslu, I've been looking a little more at the node-red-contrib-gpio side of things so I can see what I am expecting to see in terms of message structure but when I subscribe to both topics that I provide the nodebot in node red I don't see anything being published out. With the serial-mqtt should I expect to see something straight away on the publishing topic? Thanks for your help :)

monteslu, 2015-04-30:
@mcinnes01 updated mqtt-serial to allow for optional base64
If you could test that out with the microblu and it works, i'll add it to the node red nodes.

dpetican, 2015-04-30:
Still not working for me.  I tried a new token as well as changing my 
pin (used 6 and 9). I'm using the flow provided by @peterdemartini 
.Something I noticed is that my node was still online even though it had 
been physically disconnected overnight.

sharkwang, 2015-05-03:
@monteslu  I had tried the new mqtt-serial lib with microblu sketch. It worked except digitalWrite function. The debug message in digitalWriteCallback shows error pin and value.

mcinnes01, 2015-05-04:
Hi @monteslu 

I have been playing a little today with your new mqtt-serial lib, I've got the arduino connecting to my mqtt server ok, I added a little logging and an I'm alive publish message to the microblu sketch, but other than that it is unchanged. I've just been playing with your firmata example with the mqtt-serial lib.

I can see the "I'm alive message from the arduino publishing out to the 'tb' topic.

I can see the encoded message from the firmataExample.js 

But I get no console messages when running the example and I set it to turn pin 7 on and off, but get nothing physically happening on the board.

I can see on the topic the arduino is subscribed to two messages coming from the firmata example:

+Q==

and 

8Hn3

Many thanks

Andy

sharkwang, 2015-05-09:
Hi @mcinnes01 

That is correct. Because of the firmata protocol need handshake in first connect, the firmata.js send the query command like "+Q==" and "8Hn3". The microblu can't process the query, the responds lost. You need modify the firmata.js to bypass the handshake.

mcinnes01, 2015-05-10:
Hi @sharkwang 

I've been looking through the firmata.js but my knowledge of fitmata is somewhat limited and I am struggling to identify where the handshake is performed?

monteslu, 2015-05-10:
There is a skipCapabilities option that can be passed into the firmata.js controller.  Might be worth testing.  
I think there could be a problem with pins not being initialized correctly in javascript, but its been a while since I've tried that option.

sharkwang, 2015-05-11:
I had tried skipCapabilities option. It can not work.

monteslu, 2015-06-03:
have a pending PR that should help with this:
https://github.com/jgautier/firmata/pull/92/files