firmata/arduino #195: Time based pin toggles hosted on the board

fsamir, 2015-04-17:
I posted this in the mail list and Jeff suggested that I raised a feature request:

I need to be able to do the following:

board.digitalToggle(pin, HIGH, toggleInMilliseconds);

The call would set the pin to HIGH and the board would toggle it back to LOW, when the "toggleInMilliseconds" has expired.

I have been doing this in the client side, but if the client is busy, the timeout won't be honoured exactly. Also, if the client shutdown down (power outage), the pin will remain HIGH for a long time. So, I need to host the toggling logic on the board itself.

In the past, I have driven similar logic with a custom SYSEX function on the board. However, I was wondering if such a common use case isn't something that Firmata could support by default.

Does it fit in Firmata's architecture? 

Thanks,
Franklin

soundanalogous, 2015-04-17:
This is essentially a request for pulseOut right?

soundanalogous, 2015-04-17:
Also to answer your question, there is nothing like this currently in Firmata's architecture. I do however plan to add pulseIn and pulseOut support but I want to figure out a reliable way to do it that does not involve a blocking delay call.

pinocchiodrc, 2015-07-27:
I just also found this Issue, it is impossible to control the pulseOut in client side, since I use Python. I just wanna know, is it possible to do this by using a delay()?

soundanalogous, 2015-07-27:
I'll add a proposal to [firmata/protocol](https://github.com/firmata/protocol) for a pulse sequence that can be used to send one or more pulses in a single command. I drafted an initial version [here](https://github.com/rwaldron/johnny-five/issues/326#issuecomment-125040803) yesterday, but need to move it over to firmata/protocol.