firmata/arduino #197: Add INPUT_PULLUP to firmata library setPinMode function?

soundanalogous, 2015-05-17:
Copied over from [Arduino repo](https://github.com/arduino/Arduino). Originally posted by @vytronics

Is there a reason INPUT_PULLUP is not a defined mode in firmata library? When all input pins are set for INPUT mode and not connected to anything there is a tremendous amount of Tx noise for inputs changing every several milliseconds.

DomAmato, 2015-05-17:
Isn't the problem that it breaks compatibility since INPUT_PULLUP defined by arduino.h is 0x02 though i dont see why it couldn't be changed to something like FIRMATA_INPUT_PULLUP 0x0a and then handled in the code. I mean I added it in my version of firmata and haven't experienced any problems using it

soundanalogous, 2015-05-17:
Yeah that's what I was thinking of doing... assigning it to 0x0a

soundanalogous, 2015-10-25:
Planning to add this after #227 gets merged. Will have to assign FIRMATA_INPUT_PULLUP to 0x0B since 0x0A was used for Serial.

soundanalogous, 2015-11-01:
done: https://github.com/firmata/arduino/pull/235

closed 2015-11-01 by soundanalogous