firmata/arduino #207: Documentation and sample output

bitcraft, 2015-07-25:
Thanks everyone for putting together and maintaining firmata.  Its been great using it, and I look forward to new changes and features!  I'm developing a python application that needs robust arduino support (unit tested, full protocol support) and I've been developing my own python lib with integrated unit and protocol testing...since there isn't one available.

I'm wondering if there is a resource that has a breaks down of all firmata commands with sample output from an ideal firmata communication session?  Right now I am reading the docs, writing out sample output in bytes, then developing the python code to parse and build the packets.  The tests only pass if the sample output bytes can be parsed, then remade into bytes, and the inputs and outputs match.

Anyway, it would help me a great deal (and others, too!) if there was a standard file with sample outputs that could be used to testing...rather than relying on integration tests using a real board.  Is there such a thing?  Would the maintainers be interested in the files I am compiling with sample data?

Thanks!

soundanalogous, 2015-07-25:
You can find a set here that includes output from a few different Arduino boards: https://github.com/soundanalogous/Breakout/blob/master/test/core/fixtures/fixtures.js

soundanalogous, 2015-07-25:
I could convert that to a json file and host it in the firmata repo. Then others could add to it.

bitcraft, 2015-07-25:
That would be brilliant. The file you linked is almost exactly what I had
in mind. I think that it would be useful for more people. I agree,
converting it to JSON would be best.