firmata/arduino #214: Allow for multiple I2C interfaces

soundanalogous, 2015-08-23:
The current Firmata implementation (and protocol) only allows for a single I2C interface per board. For Firmata 3.0, update the protocol and implementation to allow for multiple I2C interfaces (SCA0 / SCL0, SCA1 / SCL1, etc).

rwaldron, 2015-08-24:
Excellent!

PizzaProgram, 2020-11-22:
In this case please consider to make it possible to forcibly give **unique IDs** to each I2C port!
Because linux OS is adding numbers to each one at "initialisation order".
So it can happen, that USB0 gets a bit later initialized, because the board is still booting, or other I2C ports gets activated on a Pi4 >> and the ID=**0-3** becomes suddenly **1-4** or 2 boards get's mixed and becomes **4-7**, etc... 

So it would be safer to order FIXED IDs to them when flashing the board. ID=**111-114**,  Board2: 121-124 ...

I2C_example.ino should start with lines like this:
`#define FIRMATA_I2C_ID0 111;  // Sets first I2C ports ID to override OS auto-sequence-numbering, ` 
`#define FIRMATA_I2C_ID1 112;  //   which can differ each time the device boots up.`
...

Or will be these I2C ports completely invisible for the OS and only available through "direct Firmata commands"?

soundanalogous, 2020-11-22:
Are you asking about I2C or about USB specifically? The solution to adding support for multiple I2C ports is a non-backwards compatible update to the Firmata I2C interface.

PizzaProgram, 2020-11-22:
> For Firmata 3.0, update the protocol and implementation to allow for multiple I2C interfaces (SCA0 / SCL0, SCA1 / SCL1, etc).

I'm asking (here, at this topic) only about I2C.
Maybe I'm misunderstanding something, but your plan is to add _multiple_ I2C support in Firmata 3.0?
If yes, you should consider what I wrote.

(Except, if those ports won't be reachable from outside the board firmata is installed onto.)