firmata/arduino #215: create general purpose c/c++ version of Firmata.cpp

soundanalogous, 2015-08-29:
Currently, code within Firmata.cpp directly calls many Arduino-specific functions. Create a Firmata base class that is purely c/c++ that can be hosted outside of /firmata/arduino and used therefore easily used for any c-based Firmata client.

For backwards compatibility (since many existing Arduino sketches include "Firmata.h"), the purely c/c++ version would have to be named something like FirmataBase. On the Arduino side, Firmata would either wrap or extend FirmataBase to include Arduino-specific functionality. Likewise for c-based Firmata clients or even non-Arduino architectures.

zfields, 2016-10-27:
I'm putting together a prototype for this work item.

soundanalogous, 2016-10-31:
Done via https://github.com/firmata/arduino/pull/313.

closed 2016-10-31 by soundanalogous