firmata/arduino #217: keywords issue

pierrotm777, 2015-09-06:
#######################################
# Syntax Coloring Map For Firmata
#######################################

#######################################
# Datatypes (KEYWORD1)
#######################################

Firmata	KEYWORD1	Firmata
callbackFunction	KEYWORD1	callbackFunction
systemResetCallbackFunction	KEYWORD1	systemResetCallbackFunction
stringCallbackFunction	KEYWORD1	stringCallbackFunction
sysexCallbackFunction	KEYWORD1	sysexCallbackFunction

#######################################
# Methods and Functions (KEYWORD2)
#######################################

begin						KEYWORD2
printVersion				KEYWORD2
blinkVersion				KEYWORD2
printFirmwareVersion		KEYWORD2
setFirmwareVersion			KEYWORD2
setFirmwareNameAndVersion	KEYWORD2
available					KEYWORD2
processInput				KEYWORD2
sendAnalog					KEYWORD2
sendDigital					KEYWORD2
sendDigitalPort				KEYWORD2
sendString					KEYWORD2
sendSysex					KEYWORD2
attach						KEYWORD2
detach						KEYWORD2
write						KEYWORD2
sendValueAsTwo7bitBytes		KEYWORD2
startSysex					KEYWORD2
endSysex					KEYWORD2

IS_PIN_DIGITAL				KEYWORD2
IS_PIN_ANALOG				KEYWORD2
IS_PIN_PWM					KEYWORD2
IS_PIN_SERVO				KEYWORD2
IS_PIN_I2C					KEYWORD2
IS_PIN_SPI					KEYWORD2
PIN_TO_DIGITAL				KEYWORD2
PIN_TO_ANALOG				KEYWORD2
PIN_TO_PWM					KEYWORD2
PIN_TO_SERVO				KEYWORD2

#######################################
# Constants (LITERAL1)
#######################################

MAX_DATA_BYTES				LITERAL1

STRING_DATA					LITERAL1
DIGITAL_MESSAGE				LITERAL1
ANALOG_MESSAGE				LITERAL1
REPORT_ANALOG				LITERAL1
REPORT_DIGITAL				LITERAL1
REPORT_VERSION				LITERAL1
SET_PIN_MODE				LITERAL1
SYSTEM_RESET				LITERAL1
START_SYSEX					LITERAL1
END_SYSEX					LITERAL1

ANALOG						LITERAL1
PWM							LITERAL1
SERVO						LITERAL1
SHIFT						LITERAL1
I2C							LITERAL1
ONEWIRE						LITERAL1
STEPPER						LITERAL1
ENCODER						LITERAL1
IGNORE						LITERAL1

TOTAL_PINS					LITERAL1
TOTAL_ANALOG_PINS			LITERAL1
TOTAL_DIGITAL_PINS			LITERAL1
TOTAL_PIN_MODES				LITERAL1
TOTAL_PORTS					LITERAL1
ANALOG_PORT					LITERAL1
MAX_SERVOS					LITERAL1

This is my keywords.txt version updated with good format !

soundanalogous, 2015-09-06:
What is your request here? Are you saying the existing keywords.txt file does not work due to the format or that you prefer a different format (and have added a few more keywords)?

pierrotm777, 2015-09-07:
Ho , i am sorry , in fact , the format is bad , it use space add not tab, so the colors into the IDE are bad !!!

soundanalogous, 2015-10-07:
Fixed: https://github.com/firmata/arduino/commit/f587766eb3b25eabc59554a8e190af00e063c079

closed 2015-10-07 by soundanalogous