firmata/arduino #228: I2c scanner support  feature request

ubIQio, 2015-10-13:
It would be useful if a firmata host could enumerate attached i2c devices on a firmata client.

A full scanner is not required, just the ability to detect the error from a wire.endtransmission .

cf i2cdetect, http://playground.arduino.cc/Main/I2cScanner and  http://www.gammon.com.au/i2c

Apologies if this feature already exists , but could not find it...

soundanalogous, 2015-10-13:
It does not exist but it would be low priority since you could simply run either of the above 2 examples to get the i2c address. What is the need to discover the i2c address remotely?

soundanalogous, 2015-10-13:
I could consider this for Firmata 3.0. I plan to make changes to the Firmata i2c implementation in that release in order to support the ability to have multiple i2c ports so it will be easier to add it then as the current implementation does not scale well.

ubIQio, 2015-10-13:
It would be really handy in an environment where I2C devices could be hot
plugged like Arduinos new Eslov,
https://blog.arduino.cc/2015/10/12/come-and-meet-arduino-team-in-rome/, and

Pimoroni Flotilla devices,
https://www.kickstarter.com/projects/pimoroni/flotilla-for-raspberry-pi-making-for-everyone/description

or if a given system had different build options like Ardhat
https://ardhat.com

It would be very inconvenient to unload firmata , run I2c scanner sketch
and reload firmata, just for the want of a error return message.

-- 

--

Jonathan Peace

ubIQio  *- Smart IO, everywhere <https://ubiqio.com>*

ubIQio, 2015-10-13:
Thanks Jeff.

--

Jonathan Peace

ubIQio  *- Smart IO, everywhere <https://ubiqio.com>*

scottgonzalez, 2023-10-15:
Any update on this?

pgrawehr, 2023-10-16:
@scottgonzalez This is actually an old ticket, but I think it should be closed as this is working fine. I can do a I2C bus scan from the host without any problems. See https://github.com/dotnet/iot/blob/main/src/devices/Board/I2cBusExtensions.cs#L26 for a possible client implementation.