firmata/arduino #237: enable saving board state in EEPROM as an option

soundanalogous, 2015-11-01:
This has come up several times in the past and would be worth revisiting. Past discussions were about whether or not board state should always be stored in EEPROM. I think this should be an optional feature since there are cases where a user may want to restore the state and cases where they would not. This requires more thought and planning but starting the thread now so I don't forget about it.

DomAmato, 2015-11-08:
Doesn't the EEPROM have a limited number of R/Ws though? I should probably look at the documentation before posting this but I could have sworn there is a finite number

soundanalogous, 2015-11-08:
I need to research as well. I'm not even sure if all supported boards have EEPROM or if it was only AVRs.