firmata/arduino #250: 1.6.6 Library Manager - Error downloading...

ffissore, 2015-12-14:
_From @Myriads on December 14, 2015 14:46_

Please see the attached picture:

[image]

_Copied from original issue: arduino/Arduino#4304_

ffissore, 2015-12-14:
Some notes:
1) only old firmata versions are still loaded into https://github.com/arduino-libraries/Firmata repo, recent ones come directly from https://github.com/firmata/arduino
2) There is no firmata version 2.6.2. At the time of writing, latest is 2.5.0

soundanalogous, 2015-12-15:
There was a 2.6.2 briefly but that was removed (it was a tag for a different branch). I guess if anyone used library manager for that period of 3 or 4 days when that version was included, it is was cached locally.

v2.5.0 is the latest version.

closed 2015-12-15 by soundanalogous