firmata/arduino #253: Prepend all defines with "FIRMATA_"

soundanalogous, 2015-12-19:
New boards keep breaking Firmata (and likely other libraries) by including defines and macros with the same name in their libraries. Time to fix this once and for all and prepend "FIRMATA_" to all defines. This will break many sketches unfortunately (for example if people maintain their own version of StandardFirmata or something similar and upgrade to the version of the Firmata that changes the definitions). Probably best for 3.0 with incremental patches along the way.

soundanalogous, 2015-12-19:
Another approach would be to switch to namespaced enums for Firmata 3.0

DomAmato, 2016-01-08:
Namespaced enums is probably a safer route though for consistency prepending everything with FIRMATA_ would be easier and less confusing unless there was a strong naming convention for the namespacing

soundanalogous, 2017-03-19:
No longer necessary since we now have namespacing.

closed 2017-03-19 by soundanalogous