firmata/arduino #257: ESP8266 Firmata

soundanalogous, 2015-12-24:
Add support for the ESP8266 Wi-Fi module. An arduino core for the ESP8266 is available here. Should be fairly straight forward to add Firmata support.

soundanalogous, 2015-12-24:
Looks like there is work in progress here: https://github.com/kotl/esp8266-firmata.

jacobrosenthal, 2016-02-18:
@monteslu has had me working on this today. Since esp's are in the board manager now, https://github.com/esp8266/Arduino/ StandardFirmata works just fine, you just need to add a Board definition. Heres one without analog or pwm
```
// ESP8266 generic
#elif defined(ARDUINO_ESP8266_ESP01)
#define TOTAL_ANALOG_PINS       0
#define TOTAL_PINS              17
#define VERSION_BLINK_PIN       4
#define IS_PIN_DIGITAL(p)       ((p) == 0 || (p) == 2 || (p) == 4 || (p) == 5 || (p) == 12 || (p) == 13 || (p) == 14 || (p) == 15 || (p) == 16)
#define IS_PIN_ANALOG(p)        (false)
#define IS_PIN_PWM(p)           (false)
#define IS_PIN_SERVO(p)         ((p) >= 0 && (p) < MAX_SERVOS)
#define IS_PIN_I2C(p)           (false)
#define IS_PIN_SPI(p)           (false)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p) - 17)
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         p
```

Playing with wifi now. The ESP wifi is mostly compatible, Im working on either forking the WifiStream or upstreaming changes to the ESP's wifi.h to make it 100% compat so firmata wifi should work out of the box as well.

soundanalogous, 2016-02-18:
Thanks! Looking forward to this! There are different configurations based on solder jumpers right? Maybe that is specific to Sparkfun ESP8266 boards however. Any particular ESP8266 modules you recommend? I haven't messed around with them yet but will order some.

jacobrosenthal, 2016-02-18:
https://github.com/firmata/arduino/pull/273

jacobrosenthal, 2016-02-18:
@soundanalogous I literally touched them for the first time today as a favor to @monteslu :) Just wanted to connect these dots I saw.

rwaldron, 2016-02-18:
Wow, this is awesome. 

I have 3 different ESP boards:

- SparkFun Thing https://www.sparkfun.com/products/13231
- Adafruit Huzzah https://www.adafruit.com/products/2471
- Adafruit Huzzah Feather https://www.adafruit.com/products/2821

Here are the notes I started when I first looked at the ESP8266: https://github.com/rwaldron/esp-io/blob/master/journal.md#io

monteslu, 2016-02-18:
we used the ESP8266 01 boards and connected to an ftdi chip like this to program it:
[image]

When it gets power it boots into programming mode.  

With Jacob's sketch we were able to blink pin 2 with johnny-five like this:

```javascript
var net = require('net');

var firmata = require('firmata');
var five = require("johnny-five");

var options = {   
  host: '172.22.111.111',  // IP of ESP board
  port: 3030  
};

var client = net.connect(options, function() {

  var socketClient = this;

  var io = new firmata.Board(socketClient);
  io.once('ready', function(){
      console.log('IO Ready');
      io.isReady = true;

      var board = new five.Board({io: io, repl: true});

      board.on('ready', function(){
          console.log('five ready');
          //Full Johnny-Five support here:

          var led = new five.Led(2);
          led.blink();
      });
  });

});

```

soundanalogous, 2016-03-06:
working branch is here: https://github.com/firmata/arduino/tree/esp

mwittig, 2016-03-07:
That's awesome! Thank you very much! I have managed to successfully install the firmata sketch on Wemos D1 and to run blink LED client application. I have used the following client code which is using  the "etherport-client" package I had written a couple of weeks ago to do some experiments with remote boards. 

```
var five = require("johnny-five");
var firmata = require("firmata");
var EtherPortClient = require('etherport-client').EtherPortClient;
var io = new firmata.Board(new EtherPortClient({
    host: '192.168.178.80',
    port: 3030
}));

io.once('ready', function(){
    console.log('IO Ready');
    io.isReady = true;

    var board = new five.Board({io: io, repl: true});

    board.on('ready', function(){
        console.log('five ready');

        var led = new five.Led(2);
        led.blink();
    });
});
```

I have also tried to use the J5 board directly by passing the EtherPortClient as part of the J5 board configuration (using 'port' property), but this did not work as expected: First, the program ran into a board timeout and shortly after it turned to "ready" (and the LED started blinking). May be this is because the "isReady" flag is missing somehow in this case.

soundanalogous, 2016-03-07:
J5 needs a much longer timeout when using non Serial transports.

mwittig, 2016-03-07:
Yes, possibly. This is why a created a fork of J5 which allows for passing in a timeout value as part of the board configuration. May be it makes sense to submit a PR (need to add documentation)? 

https://github.com/mwittig/johnny-five/commit/d4ced981b84d5efacf2c597490c020063a0d6259

monteslu, 2016-03-07:
+1 on that timeout option.  You should PR that.

samjulien, 2016-03-16:
I'm going to test this out with my Huzzah Feather tonight or tomorrow - is it still necessary to use the ESP working branch of firmata? Happy to provide feedback/help out if I'm able if it's useful (total n00b to this whole world so bear with me!).

soundanalogous, 2016-03-16:
yes you must still use the `esp` branch

soundanalogous, 2016-04-11:
Initial ESP8266 support has been merged into master. So far it only enables use of ESP8266 as a Wi-Fi server. Here's a simple [firmata.js client example with additional instructions](https://gist.github.com/soundanalogous/31a43d9c72ec6fbdf9631cfbe635d625)). Ability for ESP8266 to function as a Wi-Fi Client coming soon (@jnsbyr is working on it).

Tested with Adafruit HUZZAH ESP8266. Please test with other ESP boards (simply flash StandardFirmataWiFi and see wifiConfig.h for details). Make sure you have the latest Firmata master.

rahul2704, 2016-04-11:
The standard Firmata doesnt work well on Esp8266 Node MCU 0.9(ESP-12). It crashes after uploading the sketch, and simple blink sketch on johnny-five cannot connect even via usb ftdi. any clues?

soundanalogous, 2016-04-11:
You need to install StandardFirmataWiFi on the ESP8266. Make sure you have the latest version of the ESP8266 core: https://github.com/esp8266/Arduino. You need at least v2.1.0.

jnsbyr, 2016-04-11:
@rahul2704 
If you want to use the analog input you need to install the current github version. V2.1.0 has some iissues that will be fixed with the pending V2.2.0 release.

henricavalcante, 2016-04-12:
I had some troubles to compile StandardFirmataWiFi on lines:

```cpp
  if (Firmata.getPinMode(pin) == PIN_MODE_I2C && isI2CEnabled && mode != PIN_MODE_I2C) {
    // disable i2c so pins can be used for other functions
    // the following if statements should reconfigure the pins properly
    disableI2CPins(); // <<
  }
```

```cpp
  if (IS_PIN_ANALOG(pin)) {
    reportAnalogCallback(PIN_TO_ANALOG(pin), mode == PIN_MODE_ANALOG ? 1 : 0); // turn on/off reporting  // <<
  }
```

```cpp
      if (!isI2CEnabled) {
        enableI2CPins();  // <<
      }
```

After comment those lines of code I got another problem:

```
Sketch uses 236,343 bytes (54%) of program storage space. Maximum is 434,160 bytes.
Global variables use 33,184 bytes (40%) of dynamic memory, leaving 48,736 bytes for local variables. Maximum is 81,920 bytes.
Uploading 240496 bytes from to flash at 0x00000000
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
``` 

I'm using ESP8266-01

henricavalcante, 2016-04-12:
After use an external power source to ESP I got this error:
```
Sketch uses 236,343 bytes (54%) of program storage space. Maximum is 434,160 bytes.
Global variables use 33,184 bytes (40%) of dynamic memory, leaving 48,736 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x01 0x00, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

soundanalogous, 2016-04-12:
I can compile StandardFirmataWiFi on each of the ESP8266 modules supported by the [ESP8266 Arduino core](https://github.com/esp8266/Arduino) v2.1.0 except `ThaiEasyElec's ESPino` without issue using Arduino v1.6.8 on Mac OS X. The only physical board I have to test with is the Adafruit HUZZAH ESP8266 Feather.

For people having issues, please make sure you are running at least v2.1.0 of the ESP8266 Arduino core and that you're using Arduino 1.6.8 or newer (if you use the nightly build). If you are and are still having issues, specify which board you are using, what it's name is in the Arduino Boards dropdown and a link to the any info about that particular board. Also include any error messages you are seeing while compiling and/or attempting to upload the sketch.

henricavalcante, 2016-04-12:
Finally I could flash the esp8266-01 with StandardFirmataWiFi. After update everything I could compile, but I had some troubles with my Serial USB adapter on MAC, fortunately It works fine on Ubuntu.

I'm trying to connect to esp with johnny-five without success:
```js
var five = require("johnny-five");
var EtherPortClient = require('etherport-client').EtherPortClient;

var board = new five.Board({
    port: new EtherPortClient({
            host: '192.168.4.1',
            port: 3030
        }),
    timeout: 1e5,
    repl: true});

board.on('ready', function(){
    console.log('five ready');
});
````
````
1460434123414 SerialPort Connecting to host:port: 192.168.4.1:3030
1460434123418 Connected Connecting to host:port: 192.168.4.1:3030
1460434223422 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:146
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:144:17)
    at Board.log (/Users/henricavalcante/git/henricavalcante/robotjs/node_modules/johnny-five/lib/board.js:630:8)
    at Board.(anonymous function) [as error] (/Users/henricavalcante/git/henricavalcante/robotjs/node_modules/johnny-five/lib/board.js:641:14)
    at Board.<anonymous> (/Users/henricavalcante/git/henricavalcante/robotjs/node_modules/johnny-five/lib/board.js:418:14)
    at Timer.listOnTimeout (timers.js:92:15)
````

soundanalogous, 2016-04-12:
this works for me:
```javascript
var EtherPortClient = require("etherport-client").EtherPortClient;
var five = require('johnny-five');

var board = new five.Board({
  port: new EtherPortClient({
    host: "10.0.0.17",
    port: 3030
  }),
  timeout: 1e5,
  repl: false
});

board.on("ready", function() {
  console.log("READY!");
  var led = new five.Led(0);
  led.blink(500);
});
```

And blinks the red LED on the HUZZAH. However, strangely specifying pin 2 instead of 0 does not blink the blue LED, while using pin 2 with Firmata.js rather than johnny-five works without issue.

Make sure if you have `SERIAL_DEBUG` enabled in StandardFirmataWiFi that you are connected to a console or the sketch won't get past `while(!Serial) {;}` in firmataDebug.h. That could be a reason for the timeout you are experiencing.

henricavalcante, 2016-04-12:
You guys from Firmata are doing a great job, so I never had to use SERIAL_DEBUG hahaha :D

Before use the debug I was trying to use  ESP as access point which doesn't work, after use the serial debug I could see that  my ESP was trying to connect to an SSID and after configure wifiConfig.h correctly I could connect to ESP  with J5. Thanks @soundanalogous

soundanalogous, 2016-04-14:
Successfully tested boards can be added to this wiki page: https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields.

rahul2704, 2016-04-15:
I just tested on Node MCU 0.9 running esp8266 core, it works fine. added to wiki

rahul2704, 2016-04-16:
Here is an instructable. Please let me know if I omitted anything.
[http://www.instructables.com/id/ESP8266-Firmata-J5-NodeBot/](ESP8266-Firmata-J5-NodeBot)

thymjan, 2016-04-18:
I'm trying to compile StandardFirmataWiFi on my System (OSX 10.11.4, Arduino 1.6.8, board-package: esp8266 Version 2.2.0, library-package: Firmata Version 2.5.2). I've selected my board "Adafruit HUZZAH ESP8266". Now I recieve the following warnings/errors:
```
WARNUNG: Kategorie '' in der Bibliothek UIPEthernet ist ungültig und wird auf 'Uncategorized' festgelegt
In file included from /Applications/Arduino 1.8.6.app/Contents/Java/libraries/Firmata/Firmata.h:17:0,
                 from /Users/krabat/Documents/Arduino/libraries/ESP8266_Firmata/examples/StandardFirmataWiFi/StandardFirmataWiFi.ino:78:
/Applications/Arduino 1.8.6.app/Contents/Java/libraries/Firmata/Boards.h:670:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^
Mehrere Bibliotheken wurden für "Firmata.h" gefunden
 Benutzt: /Applications/Arduino 1.8.6.app/Contents/Java/libraries/Firmata
 Nicht benutzt: /Users/krabat/Documents/Arduino/libraries/arduino-configurable
 Nicht benutzt: /Users/krabat/Documents/Arduino/libraries/ESP8266_Firmata
exit status 1
Fehler beim Kompilieren für das Board Adafruit HUZZAH ESP8266. 
```
Hope this is not off-topic and I'm not to stupid for this, but I'll be happy if you can give me a hint to solve the problem.

soundanalogous, 2016-04-18:
@thymjan you need to clone or download the master branch from [firmata/arduino](https://github.com/firmata/arduino). The changes that added ESP8266 are not yet available from the version of Firmata in the Arduino Library Manager (or via the tagged v2.5.2 download).

soundanalogous, 2016-04-26:
Thanks to lots of great work by @jnsbyr, ESP8266 can now be used as either a TCP server or TCP client! Runs as TCP server by default. To run as a TCP client, simply uncomment the [SERVER_IP define](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h#L111-L114) and update to your server's IP address. Pull master for the latest.

pitchinnate, 2016-04-29:
I was trying to get Johnny-five work with an ESP8266 board. I pulled the latest Firmata code and I can flash StandardFirmata without a problem using Arduino IDE. When I try to just run a test like this:
```
var five  = require("johnny-five");
var board = new five.Board();
board.on("ready", function() {
  console.log('Board is ready !');
});
```
I get the following output:
```
1461958573646 Device(s) /dev/cu.SLAB_USBtoUART
1461958573663 Connected /dev/cu.SLAB_USBtoUART
1461958583664 Device or Firmware Error A timeout occurred while connecting to the Board.
```
Anyone have any ideas what could be causing the timeout when trying to connect?

soundanalogous, 2016-04-29:
@pitchinnate you need to pass J5 a transport. Try this:

```javascript
/*
 * Copy StandardFirmatraWiFi and enter you WiFi credentials in wifiConfig.h
 * Enable Serial debugging by uncommenting //#define SERIAL_DEBUG in StandardFirmataWiFi
 *
 * On startup (you may have to reset the ESP board because it starts up really fast.
 * View the Serial output to see the assigned IP address (if using DHCP)
 */
var five = require("johnny-five");
// by default ESP8266 is a TCP Server so you'll need a TCP client transport for J5
var EtherPortClient = require("etherport-client").EtherPortClient;
// update host to the IP address for your ESP board
var board = new five.Board(new EtherPortClient({
  host: "10.0.0.17",
  port: 3030
}));

board.on("ready", function() {
  console.log("READY!");
  var led = new five.Led(2);
  // blinks the blue LED on a HUZZAH ESP8266 board
  led.blink(500);
});
```

You may also need to set the timeout parameter to the five.Board constructor. Look up the documentation. I thinks it's simply timeout: <interval in milliseconds>.

samjulien, 2016-05-02:
I'm trying to follow the instructable above, and, this may be a dumb question, but once I uncomment the `#define SERIAL_DEBUG`, how can I read the serial output fast enough to see the startup IP address to use in johnny-five? I'm using the Adafruit Huzzah board definition (it's a Feather) and have the baud set to 115200. Once I upload the modified StandardFirmataWifi, I can't get the serial output opened fast enough. Thanks in advance.

soundanalogous, 2016-05-02:
Hit reset button on Huzzah.

samjulien, 2016-05-02:
@soundanalogous Eureka! Got your above code working on Windows 8.1 with node 5.5. Thanks!

pitchinnate, 2016-05-02:
@soundanalogous Thanks a bunch it is now working!

samjulien, 2016-05-02:
Not sure if this is a new issue or appropriate for a different medium, but is it theoretically possible for Firmata to expose the WiFi settings for configuration by something like johnny-five instead of having to configure in wifi.h?

soundanalogous, 2016-05-02:
@samjulien which WiFi settings would you like to be able to set remotely? You can't set the SSID and password since you need those before you can init J5. The only way something like that could work is if I exposed a way to configure WiFi over Serial that then internally saves the config in EEPROM (for boards that have EEPROM or an equivalent persistent storage mechanism). You'd then first have to connect via Serial with J5 to configure the WiFi then disconnect serial and reconnect over WiFi. Is possible but seems easier just to modify wifiConfig.h once.

samjulien, 2016-05-03:
@soundanalogous yep, the SSID and password are what I'm thinking. Some friends of mine are working on a tool called [flasher.js](https://github.com/thingsSDK/flasher.js), an Electron app to make it easier to flash the ESP8266. They started working on it for the purpose of flashing Espruino, but they're trying to make it modular. It occurred to me that I could compile the binary of Firmata and make a module for that in their app, but then I realized that the user needs to configure the WiFi settings. 

So, I wonder if it could be possible to 1) compile latest binary, 2) flash binary with flasher.js, 3) on success, configure WiFi? Then the user could disconnect and reconnect wirelessly with j5. What do you think?

soundanalogous, 2016-05-03:
Are you to compile and Arduino sketch file for the ESP8266 using node.js (as in running a system process). If so another option would be to use [firmata-builder](https://github.com/firmata/firmata-builder) once I finish adding wifi support for it. That will build the sketch file with arguments for the wifi config.

samjulien, 2016-05-03:
Oh, very interesting. I didn't know you could compile using Node, I'll have to look into that.

Have you had any luck with analog readings from the Feather? Last night I was trying to use a TMP36 sensor using the same circuit from the Structure/Losant kit instructions [here](https://docs.losant.com/getting-started/losant-iot-dev-kits/builder-kit/#workshop-3-temperature-sensor) but I'm getting readings from johnny-five like -50 and -58 when I use this.F or this.C.

soundanalogous, 2016-05-03:
Yes, but the analog input on the Feather is 0 - 1v so you need to use a [voltage divider](http://www.instructables.com/file/FIIS2YZI7MX2OQZ/) to get the input range to 0.0 to 1.0 volts in order to get an accurate reading.

soundanalogous, 2016-05-03:
> I didn't know you could compile using Node

I'm not sure actually which is why I was asking. If there are CLI tools for the Arduino ES8266 core it should be possible. I've never really looked into it though.

samjulien, 2016-05-03:
Okay cool, yeah I'm using a voltage divider. Must be something with my circuit or with the j5 setup. I'll look into whether there are CLI tools for the 8266. It's still the wild west out there with this chip. Thanks!

soundanalogous, 2016-05-03:
You'd have to inform J5 somewhere that the voltage range is 0.0 to 1.0 volts. I assume currently you'd also have to do the same for 3.3 vs 5v. I've never used that module however so I'm not sure how it works.

mateuszroth, 2016-05-07:
Works fine on [WeMos D1](http://www.wemos.cc/Products/d1.html), tested here: https://github.com/mateuszroth/esp8266-web-control-panel

alfredoborrero, 2016-06-25:
Hi, I am traying to connect a ESP8266 D1 Mini and i am having timeout issues.
This is a chineese version:
[http://www.aliexpress.com/item/Smart-Electronics-D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based/32628017688.html](url)

I installed latest version of the Arduino Library (Firmata Version 2.5.3) [https://github.com/firmata/arduino](https://github.com/firmata/arduino) and i was able add the parameters to **wifiConfig.h** and finally I uploaded the firmware to my ESP8266 Board (StandardFirmataWiFi.ino).

I also uncommented the line #85, so I enabled the serial debugging. And I was able to see when the board connects to my house SSID and I get a decent signal strength (-66 dBm), I can actually specify the local static IP Address or let the routher assign a Dynamic IP address from the DHCP. So its actually connecting to my WiFi Network.

The problem that I have right now is that I have timeout error when I run the Javascript file in nodeJS. And I tried with 4 different boards already. Is there anything i can do to troubleshoot this?  

`var five = require("johnny-five");
var EtherPortClient = require("etherport-client").EtherPortClient;
var board = new five.Board({
  port: new EtherPortClient({
    host: "192.168.1.64",
    port: 3030
  }),
  timeout: 1e5,
  repl: false
});

board.on("ready", function() {
  console.log("READY!");
  var led = new five.Led(2);
  led.blink(500);
});`

All I am trying to do is run the board as a Client and the server is a local host in my PC (192.168.1.64), I installed the NodeJS as well. Actually I run multiple Arduino's MEGA and Particle and I haven got any issues, for some reason I am having problems with the D1 Mini. I already read all the comments in this forum and I think I was able to troubleshoot everything but still is not working :-(

Any help would be appreciated.

Regards,

Alfredo Borrero

bkrajendra, 2016-06-29:
My board (ESP-7) after flashing only show blue led ON.. also showing following on terminal
```
ets Jan 8 2013,rst cause:2, boot mode:(3,7)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
```
Does not connect to AP. 
I'm using StandardFirmataWiFi.ino

Sny Ides.. what could be the cause.?

soundanalogous, 2016-06-29:
@bkrajendra are you using the ESP8266 Arduino board package to compile? https://github.com/esp8266/Arduino

soundanalogous, 2016-07-01:
@bkrajendra I just tested with StandardFirmataWiFi from Firmata v2.5.3 Adafruit ESP8266 Huzzah and the Arduino ESP8266 v2.3.0 board package and had no issues with either the TCP client or TCP server configuration.

I suspect either you are missing a step in the wifiConfig.h file or the ESP-7 board is for some reason not compatible. Here is the list of boards that have proven to be compatible so far: https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields.

bkrajendra, 2016-07-13:
Dear @soundanalogous  I don't think the problem is with board. ESP8266 only differs in pinouts for each board. Do you think a physical pinout could be a problem in any way? A code should generally run . Also ESP-7, ESP-12, ESP-12E and Huzzah board are almost same. There is no difference in these board which may crash or prevent to run a sketch on them.
I've tried downgrading to 2.2.0 which does not worked either 
Then tried uninstalling and cleaning everything from Arduino, ESP libraries , core and all
Done a fresh install of Arduino 1.6.10 + Fresh ESP 8266 Arduino from master + Fresh Fermata master repo.
I guess only changes required in Wificonfig is related to Line no 119 and 151 for ssid pass/key.
Which I have done .. (is anything other than this is required?)

After doing all this,
It compiles successfully with no error...
But when flashed my ESP blue (Data) Led glows continuously...
Nothing happens...

Strophe27, 2016-07-20:
Hi, 
NodeMCU 1.0 (ESP 12E) with 2.3.0 and all fresh firmata, with standardFirmataWifi, all configured, debug on, connected to the access point.
Problem : the 2 blue LEDs are ON continuously.
I send data from Max/Msp Maxuino with the mxj net.tcp.send object (instead of Serial), and I can see
```
TCP connection established
TCP connection disconnected
```
on each data sent. But nothing work : board's led (tried all pins nbr) nor servo.
Each time I try to set a pin mode in Maxuino, maxuino.js says :
`js: *** WARNING ***: For pin 2 Firmata reports this as an invalid pinMode 
`
It was just to inform you. 
I'm waiting for a ethernet shield because my goal is to make my apps with maxuino to remote 4 units by rj45. Hope it will work... 
thxs a lot guys, Firmata is... whaou... it have a long future...

soundanalogous, 2016-07-20:
@Strophe27 it's difficult for me to troubleshoot Maxuino issues and I haven't used Max/MSP in several years (don't even have a license anymore). The NodeMCU 1.0 board has been successfully tested so it's not likely an issue with Firmata you are having but rather with how Maxuino integrates with tcp. I recommend trying to get in contact with someone working with Maxuino.

Strophe27, 2016-07-20:
Thx. So, the 2 blue leds full on is a normal state ? Another strange thing is the right baud settings to see debug in serial monitor is 9600. Is that correct?

soundanalogous, 2016-07-20:
Correct, serial debug baud is 9600. The LEDs are probably on because Firmata sets all pins to output/LOW by default and I'm assuming those LEDs are wired so that LOW turns them on and HIGH turns them off.

Strophe27, 2016-07-20:
Ok. So I can see Firmata need to send back infos to maxuino.js 
I'm a very bad code reader, I try hard, but I don't understand how standardFirmataWifi send back infos, and where is it sent... 
If Firmata is in server mode, how can I tell to send infos to my Max computer ? If Firmata is in client mode, I can't receive stream from Max but I can specify the Max Ip computer.
Also I would like to debug the raw data stream input in firmata, but I can' find where to put a `DEBUG_PRINTLN` 
Maybe it isn't the right place to discuss about that beacause talking about networking (wifi + ethernet) and a client so called Maxuino... feel free to point me somewhere else

soundanalogous, 2016-07-20:
@Strophe27 file an issue on the Maxuino github repo. It doesn't look like development has been active in a while, but Chris may respond: https://github.com/maxuino/maxuino.

Strophe27, 2016-07-21:
@soundanalogous I wrote [this issue](https://github.com/maxuino/maxuino/issues/11) 2 days ago. I really need this working for my job, so I will figure it out. I wait for Ethernet shield to dig deeper into. 
I don't think Chris will do something (just an intuition, but I hope he will), so I want to contibute to Maxuino, to add Ethernet config. I know a large community of Max and Ableton Live users will be happy to have something like that.
Maybe, if I fail to reverse engineering your code I will ask you again same questions about how StandardFirmataEthernet works, maybe on a new issue, and maybe same questions above.

Back to ESP8266 (en end): my latest test are very strange... I tried different Firmatas, tried with serial, wifi etc. always LEDs on, except when serial connection is initialised (from Max), LED are off.
The exe program test you provide do not recognize the board.
But I'm done with ESP, I don't need this working for the moment, it was just for tests, waiting for shield, back on 3D design.

Strophe27, 2016-07-21:
SOLVED !
Thanks to Sadam and his awesome Max object sadam.tcpClient which is a bi-directional TCP object. (Very hard to find, since his website is down... thx googlecache...)
You were right about the two LEDs status by default.
Some more test, whith Ethernet Shield also, and I will contribute to Maxuino, and share Sadam objects to save and keep sharing them.
Thank you @soundanalogous and sorry for the noise here.

mhanuel26, 2016-10-03:
Hi bkrajendra, 

I actually have same issue here as you but with ESP-12. See #303 

Did you find a workaround? I actually brick my ESP-12 to the point that I have to remove it from PCB since bootloader was giving me problems even with already tested basic wifi example. 

http://bbs.espressif.com/viewtopic.php?f=7&t=2809&p=10045#p10045

I am afraid to do some further tests on a fresh ESP-12.

[14 more replies]

closed 2017-03-19 by soundanalogous