firmata/arduino #272: Due and Serial Protocol

WilfriedMaineult, 2016-02-09:
Using StandardFirmataPlus sketch.
I think I found a bug in the protocol implementation with Arduino Due and Serial protocol.
If i do not comment the systemResetCallback(); function at the end of setup, Serial1 does not work.
If I comment it, it works perfectly fine.
Didn't take the time to look more into this, check the PIOC registers values and so on... 
Best

soundanalogous, 2016-02-09:
I haven't had any issues with Arduino Due on Serial1, Serial2 or Serial3. I'll recheck tonight or tomorrow in case a bug was introduced with the recent Serial protocol refactor.

soundanalogous, 2016-02-09:
Also, which Firmata client library are you using?

Which version of the Arduino Due board package do you have installed?

WilfriedMaineult, 2016-02-10:
I'm using a modified version of the PyFirmata as a client. I'm pretty sure it's working since it's I only have to comment systemResetCallback() to be able to output whatever i want to output and recieve wathever I want over the serial ports.
As the bug I had was also there in the previous version.
As for the due library I use the 1.6.6 version of Arduino SAM. (Due Model R3-E for completeness).

WilfriedMaineult, 2016-02-10:
Last I checked the output of the ports with a scope, not relying on other devices...

soundanalogous, 2016-02-10:
Can you post a link to the code you added to PyFirmata to implement the Serial feature: https://github.com/firmata/protocol/blob/master/serial.md?

soundanalogous, 2016-02-11:
I just tested again to be sure:

Due board: Model R3
Due Arduino boards package: 1.6.6
Latest Firmata library (master branch), StandardFirmataPlus (January 10th, 2016 in header comment)
I'm using [firmata.js](https://github.com/jgautier/firmata) and [this script](https://gist.github.com/soundanalogous/f836998713ec0030d61b) to test.

I'm able to read a serial GPS module attached to Serial1, Serial2 or Serial3 without issue and without having to comment out `systemResetCallback()`. I don't see anything in `systemResetCallback` that would interfere with Serial functionality. This is the code in the SerialFeature class that is called when systemResetCallback is called: https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.cpp#L213-L231. SAM architecture boards don't include SoftwareSerial so the conditional block is not included. The following lines simply reset the array of ports that are tracked for `READ_CONTINUOUS` mode.

I will need to see your code in order to better diagnose the issue.

WilfriedMaineult, 2016-02-11:
Here it is : 

I only pasted the low level functions that talks to the firmata protocol. I use the Serial to talk to a lab device

class FirmataHT(A.ArduinoDue):
    def __init__(self,port='COM3'):
        super(FirmataHT, self).__init__(port,baudrate=57600,timeout=2)
        self.add_cmd_handler(SERIAL_MESSAGE,self._Handle_SpellmanResponse)
        self.add_cmd_handler(A.STRING_DATA,self._Handle_FirmataString)
        self.ConfigureSerial()
        self.SpellmanResponse = ''
        while self.bytes_available():
            self.iterate()

    def ConfigureSerial(self):
        message = self.Bit_Or(SERIAL_CONFIG,HW_SERIAL1)
        message += util.to_two_bytes(9600) + bytearray([0])
        self.send_sysex(SERIAL_MESSAGE,message)    
        time.sleep(0.1)
        while self.bytes_available():
            self.iterate()      
    
    def _Handle_SpellmanResponse(self,*data):
        data = list(data)
        if len(data)>2:
            if data[1]==2 : 
                self.SpellmanResponse=''
                data.pop(1)
            if data[1]==10 : 
                print self.SpellmanResponse
            for i in data[1:-1] :
                if len(data)>2:
                    self.SpellmanResponse+=chr(i)
            
                
    def WriteMessage(self,message):
        m = (self.Bit_Or(SERIAL_WRITE,HW_SERIAL1))
        if type(message) == type('') :
            #for car in message :
                #m = m + self._to_seven_bits(car)    
            m = m+A.util.str_to_two_byte_iter(message)
            self.send_sysex(SERIAL_MESSAGE,m)
    
    def Query(self,message, timeout=1):
        self.WriteMessage(message)
        self.ReadMessage()
        t0 =time.time()
        used = 0
        while time.time()-t0<timeout:        
            while self.bytes_available():
                self.iterate()
                self.StopReading()
		self.StopReading()

        
    def ReadMessage(self):
        message = (self.Bit_Or(SERIAL_READ,HW_SERIAL1)
                + self._to_seven_bits(0x00))
        self.send_sysex(SERIAL_MESSAGE,message)
        
    def StopReading(self):
        message = (self.Bit_Or(SERIAL_READ,HW_SERIAL1)
                + self._to_seven_bits(0x01))
        self.send_sysex(SERIAL_MESSAGE,message)

soundanalogous, 2016-02-11:
I wonder if it's possible that `self.ConfigureSerial()` is called before `systemResetCallback()` is executed in StandardFirmataPlus. This would explain the behavior since [this hack](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.cpp#L71-L73) was necessary for Due. If `systemResetCallback()` is called after that line in SerialFirmata.cpp, it will effectively disable serial for any configured ports on a Due.

The following is suspect in case it takes PyFirmata some time to establish the Serial connection, this would result in `systemResetCallback()` being called after the Serial feature is configured. Try commenting out these lines in StandardFirmataPlus: https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino#L790-L792

If that doesn't work then try commenting out only these lines in StandardFirmataPlus: https://github.com/firmata/arduino/blob/master/examples/StandardFirmataPlus/StandardFirmataPlus.ino#L739-L745.

soundanalogous, 2016-02-11:
Actually my first suggestion about lines 790-792 would not be correct since the main loop isn't called until after setup returns so there is no way `systemResetCallback()` could be called before the `SERIAL_CONFIG` message is received.

WilfriedMaineult, 2016-03-01:
Sorry about the delay, I was in a conference and did something else.
I looked at the problem today : 
Tested your second idea, and really it works when commenting the lines 739 745.
Actually, not executing the setPinModeCallback(i, OUTPUT) makes serial work properly.
If you have other test ideas...
For my part, I took the time to extract the values of the PIOA registers I could access with the Arduino Due 1.6.6 library. Maybe it can help to solve the issue. Give you the results

before the command systemResetCallback() is executed i get 
PIOA PIO_PSR
FFFFC0FC
PIOA PIO_OSR
00000000
PIOA PIO_PDSR
00000000
PIOA PIO_ODSR
00000000

after systemResetCallback I get

PIOA PIO_PSR
FFFFFCFC
PIOA PIO_OSR
0018FC80
PIOA PIO_PDSR
FFFFFDFB
PIOA PIO_ODSR
0018FC80

If i do not execute the system reset callback, thoses registers values change to the following after the  Serial configure command : 
PIOA Config - Init Serial
PIOA PIO_PSR
FFFFC4FC
PIOA PIO_OSR
00000000
PIOA PIO_PDSR
FFFFFDFB
PIOA PIO_ODSR
00000000

The same if I execute the system reset callback : 
PIOA PIO_PSR
FFFFFCFC
PIOA PIO_OSR
0018F880
PIOA PIO_PDSR
FFFFFDFB
PIOA PIO_ODSR
0018FC80

Since I'm not an expert in those chip programmation, i would take me some time to see it there a valuable information here. Hope you can find it usefull.
Wilfried

soundanalogous, 2016-03-06:
Try changing [lines 69 and 70 of SerialFirmata.cpp](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.cpp#L69-L70) to:

```c++
Firmata.setPinMode(pins.rx, PIN_MODE_IGNORE);
Firmata.setPinMode(pins.tx, PIN_MODE_IGNORE);
```

soundanalogous, 2016-03-06:
Also can you post your full code (your application code and the full PyFirmata library you are using) so I can try to reproduce the issue? I have not been able to reproduce with firmata.js, Breakout.js or johnny-five.

WilfriedMaineult, 2016-03-07:
Na it didn't work. 
I push my code to my github account tonight or so.
The thing that looks clear is that the Pio controller does not let the USART driver to control the pins.
But I don't see why... But i'm an atomic physicist, not a chip expert
Maybe I should test with another Due, to see if I can reproduce the bug with my code...

soundanalogous, 2016-11-06:
@WilfriedMaineult can you try [Firmata v2.5.4](https://github.com/firmata/arduino/releases) and see if that makes any difference regarding the issue you reported here?

closed 2020-11-24 by soundanalogous