firmata/arduino #275: String callback return half of string.

mesias, 2016-03-01:
I'm communicating firmata with Arduino, serial usb and pymata. Then couldn't understand why the strings are half size of  storedInputData buffer. It doesn't work for string communication, only receive half of on currentStringCallback.
Could you explain the reason for:
_// The string length will only be at most half the size of the
// stored input buffer so we can decode the string within the buffer._

>     case STRING_DATA:
      if (currentStringCallback) {
        byte bufferLength = (sysexBytesRead - 1) / 2;
        byte i = 1;
        byte j = 0;
        while (j < bufferLength) {
          // The string length will only be at most half the size of the
          // stored input buffer so we can decode the string within the buffer.
          storedInputData[j] = storedInputData[i];
          i++;
          storedInputData[j] += (storedInputData[i] << 7);
          i++;
          j++;
        }

soundanalogous, 2016-03-01:
Because a char is 8 bits but Firmata sends data 7 bits at a time, therefore it requires 2 bytes of the buffer for every char sent.

There may be an advantage to optimizing the encoding. If someone wants to take that on, please submit a proposal.

rwaldron, 2016-03-02:
This may be naive, but what if the protocol limited the allowed characters to ascii range?

mesias, 2016-03-02:
@soundanalogous that's explains a lot. Pymata and any other client in python doesn't have the method to send strings. Only the digital, analog, etc.
The eighth bit of byte is used to parity ? I haven't seen this thing about 7 bit's.

soundanalogous, 2016-03-02:
Firmata inherits this format from midi which it is based on. Commands are 8 bit and data is 7 bit.

Regarding limiting chars to the ascii range, that is a possibility that I've never really looked into. 

The other option is to manage the strings. You don't have to limit the string length to the buffer length, but that requires some method of copying the data from the buffer and appending it to the string you are building. In your Arduino sketch, you receive string data via a callback. See [this example](https://github.com/firmata/arduino/blob/master/examples/EchoString/EchoString.ino). The pointer `myString` passed to `stringCallback` points to the buffer containing the string. Within your callback, manage the concatenation as necessary. Once the callback returns, the data in the buffer will be lost, so this must be done before the callback returns.

soundanalogous, 2016-04-02:
Firmata String was missing from the protocol definition. Added: https://github.com/firmata/protocol/blob/master/protocol.md#string. You can actually send chars up to 14-bits.

closed 2016-04-02 by soundanalogous