firmata/arduino #281: Needs support for ATMega1284, etc, etc, etc. in Boards.h

mralexgray, 2016-04-04:
This is coming up over and over again for me.  A new day, a new MCU without Firmata support. Wonk wonk.

Additionally, this accentuates the fact that there needs to be a **simple** guide on "_Why_ `Boards.h` exists", how to extend it, and _best practices for doing so_!

Please?

Firmata should be massively extensible... isn't that the point?

soundanalogous, 2016-04-04:
The issue is time. I don't have much of it and few people contribute to this project anymore. However, anyone can submit a PR to add a new board to Boards.h. It should be (somewhat) clear by looking at other entries how to add a new one. If there is something not right I will make suggestions on how to fix it.

DomAmato, 2016-04-09:
I don't have a board to test with but just from reading the datasheet and this reference image try this
http://www.semageek.com/wp-content/uploads/2013/02/atmega1284p.png

```
#elif defined(__AVR_ATmega1284P__)
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              32 
#define VERSION_BLINK_PIN       13
#define PIN_SERIAL1_RX          8 //PD0
#define PIN_SERIAL1_TX          9 //PD1
#define PIN_SERIAL2_RX          10 //PD2
#define PIN_SERIAL2_TX          11 //PD3
#define IS_PIN_DIGITAL(p)       ((p) >= 0 && (p) < TOTAL_PINS)
#define IS_PIN_ANALOG(p)        ((p) >= 24 && (p) < TOTAL_PINS)
#define IS_PIN_PWM(p)           ((p) == 3 || (p) == 4 || (p) == 6 || (p) == 7 || (p) == 12 || (p) == 13 || (p) == 14 || (p) == 15)
#define IS_PIN_SERVO(p)         ((p) >= 0 && (p) < MAX_SERVOS)
#define IS_PIN_I2C(p)           ((p) == 16 || (p) == 17)
#define IS_PIN_SPI(p)           ((p) == SS || (p) == MOSI || (p) == MISO || (p) == SCK)
#define IS_PIN_SERIAL(p)        ((p) == 8 || (p) == 9 || (p) == 10 || (p) == 11)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        (p) - 24
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p)
```

soundanalogous, 2016-04-09:
Which Arduino-compatible board or boards use the ATmega1284?

DomAmato, 2016-04-09:
I think its a custom one. I honestly don't follow all the variants since I haven't found a need to use them for my own projects. I only posted that board for this issue but i'm not sure the onus needs to be on the firmata devs (you) to write every one of them. 

I mean it took me like 10 minutes to write that variant from a few google queries so if someone needs some random board thats custom I would hope they could figure out how to alter boards.h to make it compatible... I think its pretty straight forward tbh

DomAmato, 2016-04-09:
not sure how upto date it is but heres a list of compatible chips:
http://playground.arduino.cc/Main/ArduinoOnOtherAtmelChips