firmata/arduino #282: No BLEPeripheral.h file found when StandardFirmata being uploaded

iszufu, 2016-04-05:
I have located the latest firmata library at right place, but when I chose StandardFirmata.ino to be uploaded, there was a fatal error showing no BLEPeripheral.h file found, does that mean I need to downloaded the BLEPeripheral library offered by [sandeepmistry](https://github.com/sandeepmistry/arduino-BLEPeripheral)?

soundanalogous, 2016-04-05:
If you are trying to use StandardFirmataBLE then you need to install the [BLEPeripheral library](https://github.com/sandeepmistry/arduino-BLEPeripheral) unless you are using the Arduino 101 board. If you are using the Arduino 101 board, you need to install the Intel Curie Boars by Intel package using the Arduino Boards Manager (from the Arduino IDE: Tools -> Boards -> Boards Manager).

soundanalogous, 2016-04-05:
Hummm... I see now what you are talking about. This is a really strange error and likely a bug on the Arduino side. I'll have to figure out a work around. Until then you'll have to install the BLEPeripheral library.

soundanalogous, 2016-04-05:
fixed: https://github.com/firmata/arduino/commit/68db0550411f7f7317e8db0c65e9b4039ab22bee

closed 2016-04-05 by soundanalogous

iszufu, 2016-04-05:
Nice, now it works.