firmata/arduino #284: WiFiClient option

jnsbyr, 2016-04-09:
Currently StandardFirmataWiFi has no WiFiClient option and there was request for a contribution in https://github.com/firmata/arduino/issues/280#issuecomment-204802656.

I have compared the 3 versions of WiFiStream (WiFi, WiFi100 and ESP). They are nearly identical and they can easily be combined into a single WiFiStream class without changing the interface for WiFi and WiFi100 using defines for the ESP subset and making the appropriate changes to the StandardFirmataWiFi example. Reducing the variants first will help providing a WiFiClient option for all variants.

The new version of WiFiStream has been successfully compiled for an Arduino with WiFi and for an SP8266. I could commit and push this intermediary solution for further discussion. It still needs testing though, because I have a host application that can only be a server.

For adding the WiFiClient there are at least two options:

- adding a separate class WiFiClientStream using a copy of WiFiStream as a starting point
- integrating the WiFiClient into WiFiStream by using separate constructors for client and server

soundanalogous, 2016-04-09:
Ideally the various WiFi streams would have been combined from the beginning. The issue is however that each needs to include a different WiFi library. Arduino precompiles libraries and thus there was no way to indicate from the sketch or config.h file which specific Wi-Fi library should be included - even though that was the only difference between versions.

However since we moved the stream implementation from the .cpp to the .h file, this may have actually prevented pre compilation and including the specific Wi-Fi library in wifiConfig.h may actually work. Is that what you have in mind?

soundanalogous, 2016-04-09:
Please submit a PR with what you have so far.

soundanalogous, 2016-04-11:
@jnsbyr FYI
ESP8266 support is merged into master. Only enables using the board as a Wi-Fi server so far. Tested Wi-Fi with Adafruit HUZZAH ESP8266, Arduino MKR1000 and Arduino WiFi Shield 101 with Arduino Zero. ESP8266 and MKR1000 don't require any user configuration if using DHCP since the respective WiFi libraries are included automatically.

For the next leg of work (ability to use Wi-Fi server or client), I've created a new topic branch named `wifi-client`.

Thanks for all of your work on this so far!!

jnsbyr, 2016-04-11:
No problem, I hope that it is useful for the project. I will checkout the new branch. Thanks for all the testing you needed to do, but currently I am limited to a Huzzah ESP8266.

soundanalogous, 2016-04-11:
I recommend the MKR1000 if you are looking into getting another WiFi board. I uses the WiFi101 library so it's a good test case.

soundanalogous, 2016-04-26:
done! https://github.com/firmata/arduino/pull/292

closed 2016-04-26 by soundanalogous