firmata/arduino #288: How to make sysexCallback function pass double type parameters instead of byte type data?

iszufu, 2016-04-14:
Now, I am trying to modify the StandardFirmata to make it support a desktop-robotic arm and then I can use [snap4arduino](http://s4a.cat/snap/) to control it. When I added a new command in sysexCallback function for moveTo(double x, double y, double z) command of robotic arm, I found that the sysexCallback function can only pass byte type (byte *argv[]) parameters, but x, y and z are in double. In this case, what should I do to fix the problem, especially when the parameters are in minus?

rwaldron, 2016-04-14:
I'm not sure I understand why changes need to be made to the Firmata implementation. There are existing projects for controlling a variety of desktop robotic arms, which are already using StandardFirmata without such changes. 

The necessary controls for motors and servos are available via the `ANALOG_MESSAGE` command ([if the device is a servo, then the pin mode must be set to 4](https://github.com/firmata/protocol/blob/master/servos.md))

> moveTo(double x, double y, double z) command of robotic arm

...is just implemented in terms of three separate `ANALOG_MESSAGE` commands, with the appropriate 8-bit value or 0-180° value (depending on whether the program is controlling a motor or a servo, respectively).

iszufu, 2016-04-15:
Hmm...I think we are talking about two different solutions to control a desktop robotic arm. By using the way you said, I may need to rewrite the moveTo function in snap4arduino. But I would like to utilize a library written in c++ that provided by the arm creator. In that case, I only need to add a command in snap4arduino and give some parameters to it, then I can control it easily, just like controlling it in arduino IDE:
arm.moveTo(0, 20, -20);
The problem is the function sysexCallback (for protocol expansion) only accepting parameters in byte, but x, y and z are in double, so that it can not correctly pass the parameters. Is it possible to make sysexCallback accept parameters in double?
Thank you for your reply. If I misunderstand some parts of firmata protocol, please figure it out and help me keep moving in a right way.

iszufu, 2016-04-15:
I just find that sysexCallback(byte command, byte argc, byte *argv) function in firmata protocol is more likely to be used for passing byte data, like pin number etc, other than the 'real' parameters. So, I may have to choose the solution you implied before to make the desktop robotic arm run on snap4arduino, which will increase some work unfortunately.

rwaldron, 2016-04-15:
> But I would like to utilize a library written in c++ that provided by the arm creator.

In which case you should use any command byte not presently in use, and implement support for that library in a fork.

> The problem is the function sysexCallback (for protocol expansion) only accepting parameters in byte, but x, y and z are in double,
...
> I just realize that sysexCallback(byte command, byte argc, byte *argv) function in firmata protocol is more likely to be used for passing byte data,

Yep, looks like you've reached the conclusion that I would've explained :)

soundanalogous, 2016-04-16:
You can pass any size data using bytes simply by bit shifting. You don't need a separate function with a different signature. If you need to shift an Arduino long (which is 4 bytes) shift 8 bits and mask 4 times to break the long into 4 bytes. For an Arduino int (2 bytes) shift 8 bits and mask twice.

closed 2016-05-15 by soundanalogous