firmata/arduino #296: Firmata and APC220

dapicester, 2016-07-22:
Hi, 

I am using the [APC220 Radio Kit](http://www.dfrobot.com/wiki/index.php/APC220_Radio_Data_Module%28SKU:TEL0005%29) with my Arduino UNO, and I am able to control the board from my pc over the serial port.

Now I am trying to make it work using Firmata, so far without success.
Here is what I have tried so far.

Because the kit is set to transmit at 9600 baud, I uploaded StandardFirmata changing the line `Firmata.begin(57600)` to `Firmata.begin(9600)`. It works fine when using the arduino built-in usb (`/dev/ttyACM0` on my machine). StandardFirmataPlus also works fine.

The kit comes with a USB-serial adapter:
```
0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
```
so I tried to change the device I am connecting to (`/deb/ttyUSB1` on my machine).
I see the RX led continuously blinking on the adapter, but when I try sending some commands (e.g. turn on arduino led on pin 13), nothing happens.

If I reset the arduino (by pushing the reset button) and then try connect, then only the first command sent get executed (e.g. turn on led) but others don't (e.g. turn led off). I tried many time the sequence reset-connect-commands and it is always the same.
It seems to me that something got lost during the communication. 

I don't know if it is related, but I found a [blog post](http://www.jeffhoefs.com/2013/06/wireless-firmata-using-xbees/) by somebody having a similar issue to this but using XBees, and it seems it is due to Firmata startup sequence.

I wonder if it is actually possible to use Firmata with the APC220 and the serial adapter.

soundanalogous, 2016-07-22:
With that radio, your Arduino won't auto-reset upon establishing the serial connection like it does when you connect to a computer via usb. This is because the radio does not expose the a way to trigger reset on the Arduino board (the XBee adapter does however as described in my blog post). How much this affects you depends on which Firmata client implementation you are using (if it sends a reset command on startup you should be okay, if it does not you'll need to manually reset each time).

The issue you are experiencing may be with setting the baud to 9600, which may be too low. According to the documentation for the APC220 kit, you can increase the baud of the module up to 57600. See if that helps.

dapicester, 2016-07-25:
Thanks, I didn't realize the blog post was yours.

 I checked the client implementation I am using ([arduino_firmata](https://github.com/shokai/arduino_firmata)) and it is not sending the reset command on startup. So I send the reset command just after connection and now it seems working. I will make some more tests using something more complex that a LED blink and ensure that still works.

I haven't changed the baud of the APC220, but I may want to try that later. The blocker now is that it is possible to set the serial rate up to 56700 but the air rate can be at most 19200. Maybe I will try setting the same 19200 rate for serial and air rate in order to avoid data loss.

Kenuat, 2016-08-23:
@dapicester, I have exactly the same problem, but I'm using python library called pymata_aio. I've tried to send reset request here and there in my code, but with no luck at all :( Arduino looks totally dead (leds do not blink or anything). The scheme is correct, I've tested it with simple echo sketch. Could you please share your code change wich does the magic? So I could try do the same in pymata

soundanalogous, 2016-08-23:
@Kenuat did you try changing the baud rate in StandardFirmata from 57600 to 9600 or 19200?

Kenuat, 2016-08-23:
@soundanalogous thank you for your reply, yes, I just did (and reconfigured radio modules), behavior seems didn't changed at all.
Another thing I noticed is that Arduino sends the following string upon startup (via radio):
`щрyS*t*a*n*d*a*r*d*F*i*r*m*a*t*a*.*i*n*o`
Also I've tried to use firmata_test.exe tool, unfortunately it doesn't work as well: PC sends a few bytes and then Arduino starts sending data endlessly

soundanalogous, 2016-08-23:
The startup string is correct. One thing I suspect is the radio can't handle the capability query response which is something like 600 bytes. Does pymata_aio provide a means to bypass the capability query and pass an object mapping the pins instead?

Kenuat, 2016-08-23:
Think I can try to override pymata start method, however I'm totally sure that Arduino not even try to send these 600 bytes... It looks like it doesn't send anything at all.
How about firmata_test.exe issue? Do you have any suggestions to make it work via radio? Could these two problems be connected?

soundanalogous, 2016-08-23:
firmata_test.exe hasn't been maintained for several years

soundanalogous, 2016-08-23:
I need to know what is happening on both ends (Phython and Arduino). I don't have these radios so I'll have to walk you through adding debug statements. That will have to wait until late this evening though.

Kenuat, 2016-08-23:
Sure, I will do anything you say :) And thank you again for trying to help me with this

soundanalogous, 2016-08-24:
I have a couple of things for you to try:

1. Paste the contents of [this gist](https://gist.github.com/soundanalogous/9da533a9c3b0fc80c727027c22ee1590) into a blank sketch. Change the baud rate on line 787. Compile and run. This removes the while loop on lines 788 - 790. Let me know if that works.

2. If no luck with #1, then uncomment out `#define FIRMATA_DEBUG` (line 30), compile and upload again. In order to read the output of the debug stream, you will need listen for STRING_DATA events in your script. I'm not sure how to use it in pymata_aio. [Here's the implementation](https://github.com/MrYsLab/pymata-aio/blob/master/pymata_aio/pymata_core.py#L1632). That should hopefully print some messages, copy them and paste them in a comment.

dapicester, 2016-08-24:
@Kenuat my code is in [this gist](https://gist.github.com/dapicester/4fa59f9f39e51f54582354f50417f391). Note the `bps` options on connect, for which you need to recompile Firmata, and the `reset` command right after connecting.

MrYsLab, 2016-08-24:
@soundanalogous To try and close the loop between this issue and a similar one  for pymata-iaio https://github.com/MrYsLab/pymata-aio/issues/42 , if pymata-aio receives a STRING_DATA event, the message is sent to the console without any interaction by the user or the application. 

An issue might be the amount of time it takes the radio modules to sync after resetting the Arduino Uno. When starting pymata-aio, it first opens the serial port and then waits for 2 seconds by default before issuing a REPORT_FIRMWARE command. The delay is so that if a user resets the Arduino, pymata-aio will not try to communicate before the board is ready. This 2 second timeout is configurable if that time is not sufficient. It then issues an ANALOG_MAPPING_QUERY request so that it can configure its internal representation for the board's pinout.

If there is no communication with the board, here is what one would see on the console:
```
pymata_aio Version 2.15	Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyS0

Initializing Arduino - Please wait...
```

The com port may be reported as something else depending upon the port and OS in use.

If that banner is not seen, then there is a larger issue at hand with the apc220. If it is and there is no communication with the board, pymata-aio will wait forever for a reply. That is being changed in the next version so that if a reply is not received in 2 seconds, the application will exit.

Here is a successful connect to a Leonardo using one of my adaptations of StandardFirmata:

```
pymata_aio Version 2.15	Copyright (c) 2015-2016 Alan Yorinks All rights reserved.

Using COM Port:/dev/ttyACM0

Initializing Arduino - Please wait... 
Arduino Firmware ID: 2.5 FirmataPlus32u4.ino
Auto-discovery complete. Found 30 Digital Pins and 12 Analog Pins
```

Kenuat, 2016-08-25:
Thank you guys for helping me, @soundanalogous I think I've tried everything, however rx led at the board is not even blinking. Futhermore, I've realised that radio is not 100% functional in my setup: I can communicate with board only via tool called Serial Port Utility, while Arduino IDE port monitor is not functional. Think, I have to solve this problem first.
Thanks again, feel free to close this issue if needed, I'll try firmata again once I'll be 100% sure that everything below it is working correctly.

dapicester, 2016-08-26:
@Kenuat before you give up, let me suggest to do this: restore the default settings on the APC220 modules (so 9600 baud rate) and retry with firmata on 9600 bps, first using the cable and the with the radio. Maybe trying with a different firmata client will help (e.g. Ruby `arduino_firmata` gem and the code in my gist). I did that and it worked.

Kenuat, 2016-08-28:
@damellis thanks again, but no giving up here (my son won't let me anyway :) ) In fact, I've found the problem already:
It was in my USB-UART converter, I had to disable RTS mode programmatically to make everything to work (and set baud rate to 57600). In pyserial this can be done using Serial.setRTS(0) method (there are several init arguments and setters containing rts in name, but seems only that one makes the magic).
Anyways everything totally functional now, and I can confirm that firmata is 100% compatible with APC220 module. @soundanalogous I guess you'll want to close this issue now.

closed 2016-08-28 by soundanalogous