firmata/arduino #299: Arduino resets the previous state

aliassad, 2016-08-15:
when i send digital  message to arduino setdititalpin 8 high its work but when i send setdititalpin 9 high
the pin 8 goes low.
The arduino resets its self before sending a new message.

soundanalogous, 2016-08-15:
Are you using any particular firmata client library or are you just sending data over a uart?

soundanalogous, 2016-08-15:
Also, please paste in the exact message you are sending.

aliassad, 2016-08-15:
device.getPin(8).sevalue(1);
device.getPin(9).sevalue(1);

//the pin 9 gets high only

Im using java client liberary firmata4j

soundanalogous, 2016-08-15:
Are you setting the pin mode to OUTPUT for pins 8 and 9 before setting their value?

rwaldron, 2016-08-15:
What client library/platform is being used?

soundanalogous, 2016-08-15:
firmata4j (a Java client)

aliassad, 2016-08-17:
yes I'm setting the pins mode as output before setting their value.

soundanalogous, 2016-08-17:
I'm pretty sure the error is in the firmata4j library. There is nothing in StandardFirmata that would exhibit this behavior. I also did a quick test with the nodejs firmata.js library and this works as expected (the LEDs attached to pins 8 and 9 both light up):

```javascript
var Board = require("firmata");

Board.requestPort(function(error, port) {
  if (error) {
    console.log(error);
    return;
  }
  var board = new Board(port.comName);

  board.on("ready", function() {
    console.log("ready");
    board.pinMode(8, board.MODES.OUTPUT);
    board.pinMode(9, board.MODES.OUTPUT);

    board.digitalWrite(8, 1);
    board.digitalWrite(9, 1);
  });
});
```

soundanalogous, 2016-08-17:
issue is in firmata4j for sure: https://github.com/kurbatov/firmata4j/issues/12#issuecomment-240325691

closed 2016-08-17 by soundanalogous