firmata/arduino #302: Extra preceding bit is always sent to HardwareSerial (Serial1, Serial2, Serial3)

thorikawa, 2016-09-12:
When I use StandardFirmata on Arduino Due and connect it with a serial servo motor with Serial1 on pins 19 (RX) and 18 (TX), the servo motor does not work.

I found a issue that extra preceding LOW bit is always sent [here](https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L292) from `setPinModeCallback` function which is called from `systemResetCallback` function at the initial setup of the sketch. Call stack is as below.

```
at setPinModeCallback (https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L292)
at systemResetCallback (https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L724)
at setup (https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L771)
```

The servo motor I am using is so sensitive that this extra bit prevent it from interpreting the serial command properly. When I change the condition line [here](https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L291)
from `if (IS_PIN_DIGITAL(pin)) {` to `if (IS_PIN_DIGITAL(pin) && !IS_PIN_SERIAL(pin)) {`, everything works well. So I am really sure this line is the cause of my issue.

Do we really need [this](https://github.com/firmata/arduino/blob/2.5.3/examples/StandardFirmata/StandardFirmata.ino#L292) `digitalWrite(PIN_TO_DIGITAL(pin), LOW);` line for all digital pins including hardware serial pins at the initial setup? If not, I can send the pull request to fix it as I mentioned above.

By the way, this issue might be related with #272 which is also a issue with Arduino Due and Hardware Serial 1.

soundanalogous, 2016-09-27:
`if (IS_PIN_DIGITAL(pin) && !IS_PIN_SERIAL(pin)) {` would prevent setting the pinMode of any HW serial pins to OUTPUT if the UART is not used. This could be a common case for the Due or Mega since they each have 4 UARTs.

One approach could be:
```c++
if (Firmata.getPinMode(pin) == PIN_MODE_PWM) {
  // disable PWM
  digitalWrite(PIN_TO_DIGITAL(pin), LOW);
}
```

That way it's only called if changing a pin's mode from PWM to digital output.

soundanalogous, 2016-10-16:
PR submitted but needs testing before it can be merged: https://github.com/firmata/arduino/pull/311.

soundanalogous, 2016-10-16:
@thorikawa please checkout the [issue-302 branch](https://github.com/firmata/arduino/tree/issue-302) and report if it fixes your issue. I suspect it still could be a problem due to the RX pin being set to OUTPUT, however [this should correct that](https://github.com/firmata/arduino/blob/issue-302/utility/SerialFirmata.cpp#L75) as long as it doesn't generate a state change on the pin.

thorikawa, 2016-10-17:
@soundanalogous Thanks, I just tested against with the new branch, but did not work. I will dig into the issue more later.