firmata/arduino #303: ESP8266 problem to test Standard Firmata

mhanuel26, 2016-09-24:
Hello,

I am having an issue to make it work the StandardFirmataWiFi example on my ESP-12.

Yesterday after dealing with the example using platformio, I just brick my ESP-12, for some reason it just died the WiFi and Serial, I was still able to compile and load but it just does not make anything...

Today I started with a fresh ESP12 using Arduino IDE 1.6.12, I can upload but it just don't make anything, then I connected the serial monitor using baud = 74880 and I got this after reset

 __ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld__

I know my  ESP is working since I can load the Basic Web Server and connect to it using browser.

Do you see anything I might be missing.

soundanalogous, 2016-09-27:
It may be that the ESP-12 is not compatible with StandardFirmataWiFi. Another user had reported the same issue a while back. These are the only ESP boards that have been confirmed to work so far: https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields#esp8266.

soundanalogous, 2016-09-27:
Also, StandardFirmataWiFi uses a baud rate of 9600 to print debug info from the board. Make sure line 85 in StandardFirmataWiFi is uncommented (`//#define SERIAL_DEBUG`).

mhanuel26, 2016-09-27:
I understand that it might be some incompatibility in terms of hardware, but if it works on ESP-01 I would expect it to work on ESP-12 don't you think?
I was adding the instruction Serial.setDebugOutput(true);.

Could you point me to a python or any code to test from PC that firmataWiFi is working? 

Thank you for your comments,

soundanalogous, 2016-09-27:
Here are separate client and server examples for nodejs:

```javascript
// TCP client example
var Firmata = require("firmata").Board;
var EtherPortClient = require("etherport-client").EtherPortClient;
var board = new Firmata(new EtherPortClient({
  host: "10.0.0.20",
  port: 3030
}));

board.on("ready", function() {
  console.log("READY!");
  console.log(
    board.firmware.name + "-" +
    board.firmware.version.major + "." +
    board.firmware.version.minor
  );

  var state = 1;
  var lastVal = 0;

  this.pinMode(2, this.MODES.OUTPUT);

  setInterval(function() {
    this.digitalWrite(2, (state ^= 1));
  }.bind(this), 500);

  this.analogRead(0, function(value) {
    if (value != lastVal) {
      console.log(value);
    }
  });
});

```

```javascript
// TCP server example
var Firmata = require("firmata").Board;
var EtherPort = require("etherport");
var board = new Firmata(new EtherPort(3030));

board.on("ready", function() {
  console.log("READY!");
  console.log(
    board.firmware.name + "-" +
    board.firmware.version.major + "." +
    board.firmware.version.minor
  );

  var state = 1;
  this.pinMode(2, this.MODES.OUTPUT);

  setInterval(function() {
    this.digitalWrite(2, (state ^= 1));
  }.bind(this), 500);

  this.analogRead(0, function(value) {
    console.log(value);
  });
});
```

suside, 2017-05-03:
I had similar problem (`rst cause:4, boot mode:(3,6)`) with ESP-12. It turned out that it works perfectly fine if you provide enough amps to the board. I added old 2300mAh 3.8V smartphone battery.

soundanalogous, 2017-05-21:
Added to the Wiki, with a note to use a 500 mA or greater regulated power supply: https://github.com/firmata/arduino/wiki/WiFi-compatible-boards-and-shields#esp8266

closed 2017-05-21 by soundanalogous