firmata/arduino #304: Esp8266 i2c

d1runberg, 2016-09-29:
Hi, I'm using an esp8266 with standardfirmatwifi and j5. It works great when I blink an led, but an i2c sensor is non-responisve. I have tried with a bme280 and both the data and change event listeners never fire. I am assuming it is something to do with either the EtherPort client or firmatawifi. Any thoughts?

soundanalogous, 2016-09-30:
It shouldn't have anything to do with EtherPort since that works fine when using I2C on other Wi-Fi boards such as the Arduino MKR1000 or when using the Arduino Wi-Fi 101 shield. It could be an issue in the way I2C pins are mapped. I'll try to look into it this weekend.

d1runberg, 2016-09-30:
Thank you. I am specifically using the [SparkFun Thing Dev](https://www.sparkfun.com/products/13711) board if that helps.

soundanalogous, 2016-10-01:
I just tested using an HMC6352 digital compass with an [Adafruit Feather HUZZAH ESP8266 board](https://www.adafruit.com/product/2821) and had no issues.

Configure StandardFirmataWiIi as a TCP Server (this is the default configuration) and try this J5 program with your board (swapping the sensor with an I2C device you have in case you don't have an HMC6352):

```javascript
var EtherPortClient = require("etherport-client").EtherPortClient;
var five = require('johnny-five');

var board = new five.Board({
  port: new EtherPortClient({
    host: "10.0.0.20",
    port: 3030
  }),
  timeout: 1e5,
  repl: false
});

board.on("ready", function() {
  console.log("READY!");

  // change to the pin number your LED is connected to
  // pin 2 is SDA on the Sparkfun Thing board so don't use that
  // pin 5 is the onboard LED on the Thing
  var led = new five.Led(2);
  led.blink(500);

  var compass = new five.Compass({
    controller: "HMC6352"
  });

  compass.on("change", function() {
    console.log("heading : ", Math.floor(this.heading));
    console.log("bearing : ", this.bearing.name);
  });
});
```

soundanalogous, 2016-10-01:
You can alternately configure StandardFirmataWiFi as a TCP Client and use this J5 program:

```javascript
var EtherPort = require("etherport");
var five = require('johnny-five');

var board = new five.Board({
  port: new EtherPort(3030),
  timeout: 1e5,
  repl: false
});

board.on("ready", function() {
  console.log("READY!");

  // change to the pin number your LED is connected to
  // pin 2 is SDA on the Sparkfun Thing board so don't use that
  // pin 5 is the onboard LED on the Thing
  var led = new five.Led(2);
  led.blink(500);

  var compass = new five.Compass({
    controller: "HMC6352"
  });

  compass.on("change", function() {
    console.log("heading : ", Math.floor(this.heading));
    console.log("bearing : ", this.bearing.name);
  });
});
```

soundanalogous, 2016-10-01:
If neither of these examples work for you and you're double checked your wiring SCL to SCL and SDA to SDA, then try using the I2C device with a normal Arduino sketch. If that doesn't work, file an issue against the [Arduino esp8266 library](https://github.com/esp8266/Arduino) for the Sparkfun Thing board.

soundanalogous, 2016-10-22:
@d1runberg are you still having I2C issues with the SparkFun Thing board?

d1runberg, 2016-10-22:
Yeah. Tried a new board, different sensor...still no luck

soundanalogous, 2016-10-22:
Okay, I will probably need to order a variety of ESP8266 boards.

d1runberg, 2016-11-02:
@soundanalogous any luck with other boards? Feel free to PM/ email me if you need a SparkFun Thing.

soundanalogous, 2016-11-02:
I haven't ordered any additional board yet. I don't have any real use for them other then testing for Firmata. ESP-12 is on the top of my list since that is the board most users are reporting problems with (not just i2c issues).

soundanalogous, 2016-11-05:
@d1runberg are you using version 2.3 of the [Arduino ESP8266 library](https://github.com/esp8266/Arduino) with the Sparkfun Thing board?

bbx10, 2016-12-03:
@d1runberg  @soundanalogous 

I got a BME280 board working on a NodeMCU devkit 1.0. The I2C address of my generic BME280 board is not the default. I think the default address is set for Adafruit boards.

Most ESP-12 boards default to GPIO4=SDA and GPIO5=SCL. The SparkFun Thing uses GPIO2=SDA
and GPIO14=SCL. I am not sure if StandardFirmataWiFi handles this.

```javascript
var EtherPortClient = require("etherport-client").EtherPortClient;
var five = require('johnny-five');

var board = new five.Board({
  port: new EtherPortClient({
    host: "192.168.1.xyz",
    port: 3030
  }),
  timeout: 1e5,
  repl: false
});

board.on("ready", function() {
  console.log("board ready");

  var bme280 = new five.Multi({
    address: 0x76, // Generic BME280 boards use non-default address
    controller: "BME280"
  });

  bme280.on("data", function() {
    console.log("Thermometer");
    console.log("  celsius      : ", this.thermometer.celsius);
    console.log("  fahrenheit   : ", this.thermometer.fahrenheit);
    console.log("  kelvin       : ", this.thermometer.kelvin);
    console.log("--------------------------------------");

    console.log("Barometer");
    console.log("  pressure     : ", this.barometer.pressure);
    console.log("--------------------------------------");

    console.log("Hygrometer");
    console.log("  humidity     : ", this.hygrometer.relativeHumidity);
    console.log("--------------------------------------");
  });
});
```

soundanalogous, 2016-12-03:
Firmata will use the correct SDA and SCL pins as long as the pin variant file for the Sparkfun Thing board defines those constants appropriately. When using the [esp8266 Arduino package](https://github.com/esp8266/Arduino), these pins are mapped to [2 and 14](https://github.com/esp8266/Arduino/blob/master/variants/thing/pins_arduino.h) respectively for the Sparkfun Thing board.

rwaldron, 2016-12-04:
> The I2C address of my generic BME280 board is not the default. I think the default address is set for Adafruit boards.

From the datasheet: 

> Connecting SDO to GND results in slave address 1110110 (0x76); connection it to VDDIO results in slave address 1110111 (0x77), which is the same as BMP280’s I²C address

My guess is that Sparkfun and Adafruit opted to match the BMP280. Can you link to the generic BME280 that you're using?

bbx10, 2016-12-04:
The board in the following link looks identical to the boards I have. I probably purchased mine from a chinese website.

https://www.amazon.com/Diymall-Pressure-Temperature-Sensor-Arduino/dp/B0118XCKTG/ref=sr_1_1?ie=UTF8&qid=1480866228&sr=8-1&keywords=bme280