firmata/arduino #305: Unable to compile StandardFirmata on arduino 1.5.8

gkzsolt, 2016-10-09:
Hi,

I am trying to install Firmata using Arduino 1.5.8, but examples/StandardFirmata does not compile. After relocating several functions before they are called, I got:

../arduino-1.5.8/libraries/Firmata/utility/SerialFirmata.h:30:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>

This occurs with both releases: Firmata-2.5.3.zip and Arduino-1.6.x-Firmata-2.5.3.zip . Am I missing something ?

(as a workaround, I'd be happy to disable this, as I am not going to use SoftwareSerial anyway. But still...)

Thanks,
   Zsolt

gkzsolt, 2016-10-09:
The rejected tag in my previous comment was
`#include <SoftwareSerial.h>`

soundanalogous, 2016-10-09:
What board are you trying to compile for?

soundanalogous, 2016-10-09:
Also what OS are you using?

soundanalogous, 2016-10-09:
I just tested with Arduino 1.5.8 and it appears that Firmata is no longer compatible with older Arduino IDE versions. It actually appears to be a larger issue with the Arduino 1.5.8 IDE. I tried compiling a few other 3rd party libraries and they had the same issues. I suspect part of the issue may be that Arduino didn't support the Library Manager until Arduino 1.6.3 or 1.6.4 and the library specification changed a few times between 1.5.x and 1.6.2 or 1.6.3. If you can install a newer IDE, that should resolve the issue. In the mean time I'll try to figure out at which point Firmata (and other 3rd party libraries) stopped working.

soundanalogous, 2016-10-10:
I installed all versions of the IDE from 1.6.2 to 1.6.12 and tested with each version. It looks like you need Arduino version 1.6.6 or newer in order to compile Firmata, or you can remove Firmata from your sketchbook libraries folder which will fall back to the version of Firmata that is included with the Arduino 1.5.8 release (which is unfortunately a very old version). I have no idea what the issue is since the order of the functions is no different in the version that shipped with 1.5.8 than the present version (and as I noted I'm seeing similar issues in other 3rd party libraries). There is likely something else going on which may take me some time to track down.

gkzsolt, 2016-10-10:
Thank you. I am trying to compile for an Arduino Mega from Ubuntu 14.04. The Arduino IDE 1.6.12 solves the Firmata compilation, but unfortunately I have a quite big project which was developed under 1.5.8, and there are some other libraries which have problems with 1.6. So...

soundanalogous, 2016-10-12:
Looking over issues (including closed issues) on the Arduino repo, they had experienced a number of preprocessor issues. I suspect that is what is affecting v 1.5.8. The best I can do is reorder some functions to try to get around the issue (you really shouldn't have to forward declare functions in an Arduino .ino file). I'm not sure it's possible to get around the SoftwareSerial issue. That library has been included with all ATMega-based boards since since Arduino 1.5.x began. I'm not sure why it's not being found.

gkzsolt, 2016-10-13:
The preprocessor had to be hacked, because the .ino sketches are not regular C++ files. What I can recommend is to give forward declaration for the functions "not found", if reordering is not wanted. Regarding the SoftwareSerial issue, I'd simply put an `#include <SoftwareSerial.h>` in the sketch. Sure, it is a workaround, not the solution, but at least works.

soundanalogous, 2016-10-16:
Looking further into why SoftwareSerial.h is not being found even though the IDE includes it for the target architecture, it turns out this is because Arduino did not allow a library to reference another library (without being included in the sketch) until Arduino 1.6.6. This explains why this issue only occurs pre 1.6.6. However, I can't include SoftwareSerial in StandardFirmata.ino for all users since not all architectures support SoftwareSerial so that would crash for non AVR boards. Also SoftwareSerial is only needed if the user includes the SerialFirmata feature (but the arduino build process compiles all classes found in the project regardless). So what I'm going to do is only include SoftwareSerial in SerialFeature if the board architecture is supported (currently only AVR and ARC32) AND the IDE is v1.6.6 or greater.

soundanalogous, 2016-10-16:
https://github.com/firmata/arduino/pull/310

soundanalogous, 2016-10-16:
This is now fixed in the master branch.

closed 2016-10-16 by soundanalogous