firmata/arduino #306: Setting defined output states after upload

andreiasz, 2016-10-11:
Problem: After loading StandardFirmata into the Arduino, the GPIOs are in undefined state, destroying turnouts of my model railway. I despartely need a quick & dirty solution: 
After loading Firmata, all pins must be on digital.Output and LOW. 

Can somebody please tell me which lines of code do I need to insert and where? 

thank you!

soundanalogous, 2016-10-12:
Which Arduino board are you using?

StandardFirmata sets all pins to [OUTPUT](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L724) and [LOW](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L292) on startup.

`systemResetCallback()` is called at the end of the `setup()` function. Within `systemResetCallback` [all the non-analog pins are set to digital output](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L716-L725). When `setPinModeCallback` sets a pin to `OUTPUT`, that [pin value is set to LOW](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L292).

However, if you are using an ATMega-based board, Arduino defaults all pins to INPUT mode so on power up, all pins will switch from INPUT to OUTPUT and this may be causing the issue you are experiencing. Or if you are using an non ATMega-based board it could be that the GPIO pins are actually in an undefined state. Firmata defines a state for them, but this very short delay on initial boot may be enough to cause an issue.

gkzsolt, 2016-10-15:
Yes, StandardFirmata sets all pins to OUTPUT and LOW on startup, what came as a shock for me, because my circuit has also pins connected as INPUT to other controllers. When I changed my sketch to  StandardFirmata, some of them got short-circuited by colliding with HIGH logic level from the other part, so it was a miracle that the circuits did not get damaged. 
All pins should default to INPUT mode on startup, maybe with pull-ups connected. This is the only mode which is safe in every hardware situation. Who's idea was to set all pins to OUTPUT mode?

(Btw, is there a forum for discussion, or I should open an issue for any question I have)

gkzsolt, 2016-10-15:
andreiasz, I use the following to set up pin modes and values:
`#define PIN_RELAY_4 17
setPinModeCallback( PIN_RELAY_4, OUTPUT );
setPinValueCallback( PIN_RELAY_4, HIGH );`
for example to set pin 17 as OUTPUT and to HIGH. (You said you need OUTPUT LOW, which really _is_ what StandardFirmata currently provides...)
I don't know if this is correct or there is a simpler way. The Firmata class also has setPinMode and setPinState functions, but I'm afraid they are not enough for StandardFirmata to function correctly (?)

soundanalogous, 2016-10-15:
Please see my proposed solutions here: https://github.com/firmata/arduino/issues/308.