firmata/arduino #309: Add Teensy 3.5 and 3.6 to Boards.h

balda, 2016-10-16:
New Teensy 3.5 and 3.6 are not supported (all other Teensy boards are).

`Please edit Boards.h with a hardware abstraction for this board`.

soundanalogous, 2016-10-16:
I can add them once they are supported by Teensyduino, but I don't think they are yet. At least no info is published: https://www.pjrc.com/teensy/teensyduino.html.

soundanalogous, 2016-10-16:
A PR is also encouraged here if someone already has either of these boards.

balda, 2016-10-16:
Teensy 3.5 and 3.6 are supported, but still in beta: https://forum.pjrc.com/threads/36756-Teensyduino-1-30-Beta-4-Available

I have both, but i'm not experimented enough to code board info. But i can make some test :)

soundanalogous, 2016-10-17:
Here's a first pass: https://github.com/firmata/arduino/pull/312

I'm only able to verify that it compiles for Teensy 3.5 and 3.6 (in Arduino 1.6.11 with the latest teensyduino 1.30 beta 4). Please test with actual hardware and let me know if you encounter any issues. Try reading analog input on some of the higher number pins like A21 and A22 and reading or digital input or writing digital output on a selection of digital pins between 24 and 39.

soundanalogous, 2016-10-22:
done

closed 2016-10-22 by soundanalogous