firmata/arduino #314: Help wanted: reconnect / connection error handling

jsiegmund, 2016-11-01:
Filing an issue for some guidance as I couldn't find any info on this topic. I noticed the help-wanted tag so hope this is allright :) I am using firmata loaded onto an Arduino Nano to connect with a Raspberry Pi over bluetooth. I've got a bluetooth shield connected to RX/TX and use the firmata protocol in combination with the firmata .NET SDK (Pi is running Win10IoT). All works well! Until is doesn't... Bluetooth is error prone and for whatever reason a module may be disconnected. This is expected and I've got code on the host (raspberry) to handle this, detect & reconnect, but this doesn't always work.

In some cases it seems the connection cannot be reestablished for whatever reason. Recycling the Arduino fixes this; usually I just pull the plug and plug it in again. So my guess is that on either sides the connection is left in a unusable state. Recycling the process on the Pi doesn't fix this, so my focus shifted to the Arduino.

Because I couldn't find too much detail on this, I am left wondering: 
- Should I be able to reconnect, no matter how the connection was broken? 
- Is this something that firmata can/should deal with or is it a problem of some underlying library like Serial? 
- Is there any documentation out there guiding on how to cope with connection failures? 
- An example of resilient connections would be awesome of course, but I will do with less

Any sort of additional information would be welcome!

soundanalogous, 2016-11-02:
I believe this may be a problem specifically with the ATmega32u4 (Nano, Leonardo, etc). Those boards don't auto reset like many other Arduino boards which will reset automatically when a serial connection is established. You can send a soft reset using the SYSTEM_RESET command (0xFF). You may also need to send a REPORT_VERSION or REPORT_FIRMWARE command depending on what the firmata .NET SDK expects as a handshake upon startup.

There is currently no way to detect connection status on the firmware side when using a serial connection. A [host connection callback has been added for WiFi](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/StandardFirmataWiFi.ino#L844-L860) but it doesn't do anything yet. It lays the groundwork for this functionality for network transports and for BLE. I don't have any plans to add something like this for Serial any time soon however as I'm not exactly sure how to detect a serial disconnection without some sort of a periodic ping.

jsiegmund, 2016-11-02:
Ok, that's clear info, thanks! The SYSTEM_RESET is cool but if there's no connection to begin with I wouldn't know how exactly to send it. In the meantime I found that sometimes the blame is on the host side as well so I'm going to inspect what I can do there to make the experience at least a bit better. 

Thanks for the info!

closed 2016-11-02 by jsiegmund