firmata/arduino #315: Add ESP32 to board defs

d1runberg, 2016-11-02:
It would be awesome to add ESP32 to the board defs for standardfirmatawifi/ bluetooth! The GPIO capability is much better than the 8266

soundanalogous, 2016-11-02:
Do you have an ESP32 board? I don't and need someone to verify the defs if I add them.

d1runberg, 2016-11-02:
Not yet, but I will probably have one in about a week ;)

soundanalogous, 2016-11-02:
Also I'm not sure if anyone has added Arduino support for it yet. Will need to wait until it's available in the Arduino Board Manager.

d1runberg, 2016-11-03:
Here is the Arduino core...

https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide?_ga=1.116824388.33505106.1471290985#installing-the-esp32-arduino-core

Derek Runberg
Regional Education Developer
Sparkfun Electronics Inc.

Boulder, CO.
Cell: 971-344-2355
derek.runberg@sparkfun.com
http://www.sparkfun.com
http://learn.sparkfun.com

soundanalogous, 2016-11-03:
Thanks! Hadn't realized you worked at Sparkfun.

d1runberg, 2016-11-03:
;)

Derek Runberg
Regional Education Developer
Sparkfun Electronics Inc.

Boulder, CO.
Cell: 971-344-2355
derek.runberg@sparkfun.com
http://www.sparkfun.com
http://learn.sparkfun.com

deadprogram, 2017-01-13:
Thank for working on this.

I see that the tools install themselves into the Arduino environment by following the Sparkfun site instructions. What, if anything, else needs to be done to run WiFi Firmata on the ESP32?

soundanalogous, 2017-01-13:
A lot needs to be done. An entry for the ESP32 is needed in [Boards.h](https://github.com/firmata/arduino/blob/master/Boards.h). An entry must also be added to [wifiConfig.h](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h) and any pins that need to be ignored must be specified. Changes may need to be made to [WiFiStream.h](https://github.com/firmata/arduino/blob/master/utility/WiFiStream.h) to handle any ESP32-specific behavior that is not 100% aligned with the Arduino Wi-Fi 101 library behavior, etc.

I recently picked up an ESP32 Thing board so I'll be adding this at some point in the coming weeks.

deadprogram, 2017-05-17:
Hi @soundanalogous anything happen on this, or anything I can do to help out with it?

soundanalogous, 2017-05-17:
I haven't had any time to work on it. Adding UART support would only take a few minutes, but adding Wi-Fi and BLE will take a lot longer. I don't even thing the ESP32 Thing has an Arduino BLE library yet. It does have Wi-Fi though.

deadprogram, 2017-05-17:
Thanks for the reply @soundanalogous 

If it is literally that easy, perhaps you could add the UART support, and then others like myself can work on adding WiFi support using https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi

It does appear that there is also now BLE support as well https://github.com/espressif/arduino-esp32/tree/master/libraries/SimpleBLE

soundanalogous, 2017-05-17:
That would be great if you want to take on Wi-Fi support! I'll see if I can find time to add basic support this weekend.

I've looked at SimpleBLE, but unfortunately it's not comprehensive enough. I'm hoping ESP32 support is added to the [BLEPeripheral library](https://github.com/sandeepmistry/arduino-BLEPeripheral) at some point since that already works with Firmata.

soundanalogous, 2017-05-20:
I looked into this today. It turns out to be not so simple because the arduino-esp32 library does not yet support `analogWrite` or the Servo library which are both expected by StandardFirmata (and variants). I don't want to add an #ifdef to StandardFirmata just for this architecture either.

I could probably add the board to ConfigurableFirmata, but a user would have to know to exclude the `analogWrite` and `Servo` features in order to avoid compiler errors.

soundanalogous, 2017-05-20:
The  esp32 HAL also redefines the standard `INPUT` and `OUTPUT` constants which is annoying. It's a relatively easy fix in Firmata however.

deadprogram, 2017-05-21:
Requiring a specific ConfigurableFirmata is not a problem for me personally, and certainly is an interim step until those other entrypoints are supported by arduino-esp. As long as it is documented, others will be able to avoid pitfalls with that.

soundanalogous, 2017-05-22:
This board is not proving to be very Firmata friendly. Turns out it spits out a large string of text on startup and upon establishing a new Serial connection:

```
⸮⸮⸮ ⸮   ⸮ ⸮⸮⸮⸮⸮ ⸮⸮             ⸮   ⸮        ⸮        ⸮ ⸮⸮⸮ ⸮⸮ ⸮⸮  ⸮ ⸮⸮  ⸮   ⸮⸮ ⸮ ⸮⸮ ⸮    ⸮⸮     ⸮  ⸮⸮⸮ ⸮⸮⸮⸮⸮⸮⸮ ⸮  ⸮⸮⸮ ⸮  ⸮⸮⸮⸮  ⸮⸮⸮⸮ ⸮⸮ ⸮⸮⸮⸮ ⸮ ⸮⸮⸮⸮ ⸮  ⸮   ⸮⸮⸮⸮⸮⸮⸮ ⸮⸮ ⸮    ⸮⸮  ⸮     ⸮ ⸮ ⸮⸮ ⸮⸮⸮  ⸮⸮⸮⸮  ⸮⸮ ⸮⸮ ⸮⸮ ⸮⸮⸮⸮⸮⸮  ⸮  ⸮⸮ ⸮⸮ ⸮⸮  ⸮⸮⸮⸮  ⸮⸮ ⸮⸮ ⸮ ⸮⸮⸮⸮ ⸮ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
```

A firmata client would need to interpret and ignore this data in order for this board to work over a Serial transport. This won't affect Wi-Fi however so that is still a possible option, but overall this board seems like a poor choice for Firmata applications.

soundanalogous, 2017-05-29:
I have ConfigurableFirmata branch with ESP32 support that sorta works for Wi-Fi: https://github.com/firmata/ConfigurableFirmata/pull/71. It's far from stable however. Needs testing and I could use some help improving it... especially in figuring out a buffering scheme that will likely lead to a larger structural change in using non-Serial/USB transports with Firmata.

aycgit, 2017-07-18:
any progress? im thinking of picking up an esp32 soon

soundanalogous, 2017-07-18:
I would not recommend purchasing an ESP32 board specifically for use with Firmata any time soon. Buy an ESP8266 board instead.

aycgit, 2017-07-18:
okay, now i have to decide to get bluetooth or firmata. Anyway, thanks for your work on this awesome project.

soundanalogous, 2017-07-18:
BLE is supported for Firmata on the Arduino 101 and the RedBear BLE Nano (with some minor changes). However, that said, BLE does not work well with Firmata because Firmata is stream-based and BLE is not designed for large data transfers.

soundanalogous, 2017-07-18:
I should add that how well BLE works with Firmata really depends on your application. It actually works well if you just need data from a single sensor or need to control a single actuator. It doesn't work well for general purpose control of a microcontroller where you need to use various digital and analog signals simultaneously.

ttww, 2018-01-05:
Ups, I wanted more pins and changed to esp32, but firmata is not supported... :-(
Any news/infos/plans on this issue?

soundanalogous, 2018-01-05:
The biggest blocker is that the Arduino `analogWrite` method is not yet supported on the esp32.

monteslu, 2018-01-05:
I'm using ledcwrite as a workaround until we get analogwrite:  https://github.com/monteslu/ble-io/blob/master/arduino/esp32/esp32.ino#L229

Might still be helpful to have the board definition so we know what pins can do what in the meantime

ttww, 2018-01-05:
Thanks for that tip :-)

draeron, 2019-08-11:
does anyone has a functioning fork with the ESP32 working, i don't need `analogWrite` so this ain't a problem for me.

soundanalogous, 2019-08-13:
@draeron you can try [this branch](https://github.com/firmata/ConfigurableFirmata/tree/esp32-thing) and see my notes about it [here](https://github.com/firmata/ConfigurableFirmata/pull/71).

draeron, 2019-08-13:
thx i'll be sure to check it out.

T0T4R4, 2020-07-26:
Firmata provided in IDE 1.8.13 still seems not to be working on bpi:uno32 , any plan to solve this issue ?

NANDLAB, 2021-11-02:
I would also appreciate support for ESP32.

> https://github.com/firmata/ConfigurableFirmata does have support for the ESP32. I didn't test it extensively, but the basic features seem to work.

@pgrawehr, did you get it to work? I tried compiling a ConfigurableFirmata project, which I downloaded from [firmatabuilder](http://firmatabuilder.com/) with connection over serial and only digital input and output features, but I get this error:
```
In file included from /home/dan/Arduino/libraries/ConfigurableFirmata/src/ConfigurableFirmata.h:18,
                 from /home/dan/Arduino/MyConfigurableFirmata/MyConfigurableFirmata.ino:6:
/home/dan/Arduino/libraries/ConfigurableFirmata/src/utility/Boards.h:711:2: error: #error "Please edit Boards.h with a hardware abstraction for this board"
 #error "Please edit Boards.h with a hardware abstraction for this board"
  ^~~~~
Using library ConfigurableFirmata at version 2.10.1 in folder: /home/dan/Arduino/libraries/ConfigurableFirmata 
exit status 1
Error compiling for board ESP32 Dev Module.
```
My board is "ESP32 Dev Module".

pgrawehr, 2021-11-03:
@NANDLAB The officially released version doesn't support the new features, you need to check out master manually. (I need yet to figure out how to create a new release). 

Do the following: 
```
cd ~/Arduino/libraries
rm -rd ConfigurableFirmata
git clone https://github.com/firmata/ConfigurableFirmata
```

This checks out the latest version that does include ESP32 support. I'm not sure the scripts built with firmatabuilder are still compatible with it. If you get errors, start with `examples/ConfigurableFirmata/ConfigurableFirmata.ino` instead (there's a bunch of `#ifdef`s at the beginning to enable/disable modules as required).

hgmarques, 2022-07-20:
Hi guys...

does anyone know if there is any progress on using StandardFirmataWifi with a ESP32 board?

Thanks in advance,

pgrawehr, 2022-07-21:
@hgmarques No, there's currently no development activity on StandardFirmata (except critical bug fixes). 

Development is focusing on ConfigurableFirmata, which _does_ have ESP32 Wifi support. Just check out ConfigurableFirmata master, open `ConfigurableFirmata.ino` and edit the wifi config options at the start of the file.

hgmarques, 2022-07-21:
OK. Thanks a lot, that was very useful. 

I installed the ConfigurableFirmata but got a compilation error indicating that I needed to add my board (ESP32 Thing Plus) to Boards.h (which I have done in the past for other boards in StandardFirmata). However, in Boards.h of ConfigurableFirmata I don't find anything option resembling ESP32 boards. Any advice? 

(Thanks in advance)

pgrawehr, 2022-07-21:
@hgmarques ESP32 is defined in `src/utility/Boards.h` line 701. At least in the version I have...

pgrawehr, 2022-08-01:
@hgmarques Where you able to get it running? Or are there still problems?

hgmarques, 2022-08-02:
Hi Patrick thank you for your prompt reply. I don't have the board with me at the moment but I'll get back to you as soon as I do.

hgmarques, 2022-09-11:
So... I got my hands on a ESP32-S2 Thing Plus to test the ConfigurableFirmataWifi but I was unsuccessful in getting it running.

**1. I get the same results when compiling for boards**:
. Adafruit ESP32 Feather (which I've used successfully in the past for a previous version of the board), and
. Sparkfun ESP32-S2 Thing Plus. I get the same results (see below)
(I can see the ESP32 macro defined in the ConfigurableFirmataWifi)

**2. I'm using WIFI_101** (I'm not sure if this is the right choice). 

**3. I installed the ESP32Servo library and I modified references to Servo.h to ESP32Servo.h in the ConfigurableFirmataWifi and the ServoFirmata files to have the servo libraries compatible with ESP32.** This avoids the following error:

_WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed, mbed_nano, mbed_portenta, mbed_rp2040 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from C:\Users\hgmar\AppData\Local\Temp\arduino_modified_sketch_415765\ConfigurableFirmataWiFi.ino:345:
C:\Program Files (x86)\Arduino\libraries\Servo\src/Servo.h:77:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."_

**4. Once I get past the simpler errors I get these errors, which seem already related to the ConfigurableFirmata. Any ideas? (thanks in advance)**

In file included from C:\Users\hgmar\Documents\Arduino\libraries\ConfigurableFirmata\src/OneWireFirmata.h:19,
                 from C:\Users\hgmar\AppData\Local\Temp\arduino_modified_sketch_182788\ConfigurableFirmataWiFi.ino:357:
C:\Users\hgmar\Documents\Arduino\libraries\ConfigurableFirmata\src/utility/OneWire.h:245:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
 #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
  ^~~~~~~
In file included from C:\Users\hgmar\Documents\Arduino\libraries\WiFi101\src/WiFiClient.h:28,
                 from C:\Users\hgmar\Documents\Arduino\libraries\WiFi101\src/WiFi101.h:35,
                 from C:\Users\hgmar\AppData\Local\Temp\arduino_modified_sketch_182788\ConfigurableFirmataWiFi.ino:131:
C:\Users\hgmar\Documents\Arduino\libraries\WiFi101\src/socket/include/socket.h:1839:15: error: conflicting declaration of C function 'sint8 close(SOCKET)'
 NMI_API sint8 close(SOCKET sock);
               ^~~~~
In file included from C:\Users\hgmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3/tools/sdk/esp32s2/include/newlib/platform_include/sys/unistd.h:23,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\sys-include\unistd.h:4,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\sys-include\pthread.h:25,
                 from C:\Users\hgmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3/tools/sdk/esp32s2/include/newlib/platform_include/pthread.h:21,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\xtensa-esp32s2-elf\no-rtti\bits\gthr-default.h:48,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\xtensa-esp32s2-elf\no-rtti\bits\gthr.h:151,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\ext\atomicity.h:35,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\bits\basic_string.h:39,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\string:52,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\array:39,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\hgmar\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32s2-elf\include\c++\8.4.0\functional:54,
                 from C:\Users\hgmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/HardwareSerial.h:49,
                 from C:\Users\hgmar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/Arduino.h:167,
                 from sketch\ConfigurableFirmataWiFi.ino.cpp:1:
[... cut ...]
          ^~~~~~
C:\Users\hgmar\Documents\Arduino\libraries\ConfigurableFirmata\src/I2CFirmata.h:58:10: note:   candidate expects 1 argument, 0 provided
ConfigurableFirmataWiFi:682:14: error: 'class AccelStepperFirmata' has no member named 'update'
 accelStepper.update();
              ^~~~~~
ConfigurableFirmataWiFi:685:10: error: 'class SerialFirmata' has no member named 'update'
   serial.update();
          ^~~~~~
exit status 1
no matching function for call to 'DigitalInputFirmata::report()'

pgrawehr, 2022-09-11:
@hgmarques 

To begin with, use ConfigurableFirmata.ino, not the Wifi variant, that one is outdated and should probably be deleted. The "standard" file `ConfigurableFirmata.ino` contains all necessary declarations to run with Wifi as well (just edit the declarations at the beginning or create a `WifiConfig.h` file). This should solve many of your problems already. 

The default Servo library is unsupported on the ESP32 as you have found out. I have not done anything with servo motors, so I don't know whether the ESP32 Servo library works correctly. 

The message
```
#warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
```
can be ignored. It should work anyway (and possibly you don't even need one wire support, but the library is required for I2C support as well).

hgmarques, 2022-09-11:
Very nice!! It compiles well... Thanks a lot! I would not have done it without your help.

Now, I get this error when running in the monitor (below)... so it is not clear to me how whether the WIFI connection has actually been established... Any ideas? If not, I'll try to debug from here ..., ,)

_Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002c45f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28cc
entry 0x4004c18c
E (6597) gpio: gpio_set_level(226): GPIO output gpio_num error
E (6598) gpio: gpio_set_level(226): GPIO output gpio_num error_

Thanks again...

hgmarques, 2022-09-11:
Perfect! It works! The connection is established (below).

I'm actually not even trying to connect with device from the outside yet... I'll check the error and I'll get back to you once I get it sorted... ;)

Thanks a lot...

_Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40027bba
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28cc
entry 0x4004c18c
[   620][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=3030, ...)
[   634][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   669][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   670][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[   672][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   737][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: MEO-443C10, BSSID: 00:06:91:44:3c:10, Channel: 1, Auth: WPA2_PSK
[   739][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   780][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.1.133
[   780][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   783][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.1.133, MASK: 255.255.255.0, GW: 192.168.1.254
E (4218) gpio: gpio_set_level(226): GPIO output gpio_num error
[  4669][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
E (4219) gpio: gpio_set_level(226): GPIO output gpio_num error
[  4675][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
[  4681][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected_

jflamy, 2022-11-10:
@hgmarques have you made more progress ?  Am interested in understanding how you got communication going over a virtual serial port (or if there is a better way).

pgrawehr, 2022-11-11:
@hgmarques What do you mean by "virtual serial port"? The ESP32 can communicate over the standard USB-to-Serial connection (the same that is used for programming) or trough Wifi.

hgmarques, 2022-11-11:
@jflamy Unfortunately I have not... but I concur with pgrawehr -- I was trying to use plain WiFi... 

For now, I'm sticking with the old MKR1000, which I think it has even be discontinued, but is the only one I have been able to use Firmata on... :( I'll try to come back to the ESP32 later in my project.

pgrawehr, 2022-11-11:
@hgmarques I strongly recommend to use ConfigurableFirmata for ESP32. It has been (quite troughfully) tested and works just fine for me.

hgmarques, 2022-11-11:
Yes... I agree... I actually meant ConfigurableFirmata. ;)

pgrawehr, 2023-03-29:
ESP32 is supported for ConfigurableFirmata. Since it has more than enough memory, a backport is probably not worth the effort. Therefore closing this ticket.

closed 2023-03-29 by pgrawehr