firmata/arduino #322: older avr-gcc compiler doesn't handle array initialization in constructor list

soundanalogous, 2016-11-07:
I'm trying maintain Firmata back to Arduino v1.0.6. I noticed the compiler doesn't know how to handle array initialization in the FirmataParser constructor list. I get the following error:

```
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp: In constructor 'FirmataParser::FirmataParser()':
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp:32: error: expected `(' before '{' token
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp:32: error: expected `;' before '}' token
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp:32: warning: statement has no effect
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp: At global scope:
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp:32: error: mixing declarations and function-definitions is forbidden
/Applications/arduino-1.0.6/Arduino.app/Contents/Resources/Java/libraries/Firmata/FirmataParser.cpp:33: error: expected constructor, destructor, or type conversion before '(' token
```

Worth even trying to find a work around to maintain support for older Arduino versions?

soundanalogous, 2016-11-07:
@zfields please take a look at this.

soundanalogous, 2016-11-07:
Fails on `storedInputData{0},` in the following:

```c++
FirmataParser::FirmataParser()
:
    executeMultiByteCommand(0),
    multiByteChannel(0),
    storedInputData{0},
    waitForData(0),
    parsingSysex(false),
    sysexBytesRead(0),
    currentAnalogCallback((callbackFunction)NULL),
    currentDigitalCallback((callbackFunction)NULL),
    currentReportAnalogCallback((callbackFunction)NULL),
    currentReportDigitalCallback((callbackFunction)NULL),
    currentPinModeCallback((callbackFunction)NULL),
    currentPinValueCallback((callbackFunction)NULL),
    currentReportFirmwareCallback((systemCallbackFunction)NULL),
    currentReportVersionCallback((systemCallbackFunction)NULL),
    currentSystemResetCallback((systemCallbackFunction)NULL),
    currentStringCallback((stringCallbackFunction)NULL),
    currentSysexCallback((sysexCallbackFunction)NULL)
{
}
```

zfields, 2016-11-07:
This is an easy fix, I'll have it out tonight.

zfields, 2016-11-07:
I have two options for you.

It is unnecessary to initialize this, I was just doing it for good measure. We can eliminate it completely, or we can put the following in the constructor:
```c++
    size_t i = (MAX_DATA_BYTES - 1);
    for (; i >= 0 ; --i) {
        storedInputData[i] = 0;
    }
```
Let me know how you want to handle it.

zfields, 2016-11-07:
I've decided that I would just delete the offending code, as it is unnecessary processing to set it to zero (not to mention slightly OCD-esque). However, I'll leave the final call up to you.

soundanalogous, 2016-11-07:
lets just delete the offending code

zfields, 2016-11-07:
I'll have a PR out momentarily.

zfields, 2016-11-07:
#323 Is a one liner, ready for review

closed 2016-11-07 by soundanalogous