firmata/arduino #328: [QUESTION] What is the motivation for `MAX_DATA_BYTES`?

zfields, 2016-11-11:
`MAX_DATA_BYTES` is the only reason I need to include `FirmataConstants.h` in `FirmataParser.h`.

Depending on the answer to a few questions, I would like to offer provide a replacement.
1. Does `MAX_DATA_BYTES` play a larger role than defining the size of the state?
2. Does the array `storedInputData` need to be on the stack, or was this done out of convenience?
3. Would a `char *` stored on the heap, using `realloc()` be an acceptable substitute?

soundanalogous, 2016-11-11:
It's legacy from when Firmata only worked over a Serial transport and there were only a couple of Arduino boards. It should really be both platform and transport dependent.

soundanalogous, 2016-11-11:
I believe `realloc()` is dangerous for boards with limited ram (as in 2.0 and 2.5k boards) as there may be risk of memory fragmentation. It could be used if the RAM is high enough however. I think there is a way to check RAM size.

zfields, 2016-11-11:
Does `alloca()` work on those boards? That may be a way to respond to the data, and keep the memory intact.

zfields, 2016-11-11:
Scratch that, even if `alloca()` is supported by the Arduino tool chains, it still isn't what we want. I understand the points you've made. I'll try and find a more elegant solution.

closed 2016-12-23 by soundanalogous