firmata/arduino #329: `FirmataParser::detach` usage is misleading

zfields, 2016-11-12:
`FirmataParser::detach` checks for `START_SYSEX`, but `FirmataParser::attach` will accept any variable as `command` argument.

To be consistent, it feels like `FirmataParser::attach` should require `START_SYSEX` or do nothing.

soundanalogous, 2016-11-12:
> FirmataParser::detach only checks for START_SYSEX

I'm not sure exactly what you mean here. FirmataParser::detach currently checks several cases, not just `START_SYSEX`: https://github.com/firmata/arduino/blob/master/FirmataParser.cpp#L260-L273.

There is a chance of error however in that the user could pass an unsupported callback that. We could break up detach into several functions like attach or add even more cases and throw an error on default since then it could be setup to only catch unrecognized callback interfaces.

zfields, 2016-11-12:
That was my intention. It is confusing, because you a single callback for sysex, which makes sense, but you can attach it in a way that wouldn't allow you to detach with the same `command` byte.