firmata/arduino #331: unit tests for FirmataParser

soundanalogous, 2016-11-12:
Now that the parsing functionality has been extracted from the Arduino-specific Firmata.cpp implementation, it would be useful to add unit tests for FirmataParser. Existing tests use ArduinoUnit which runs on the board (which really puts them more in the functional test category), but tests for FirmataParser could simply run on the PC. I'm not familiar with general C++ unit testing libraries (or native c++ testing support for that matter). @zfields curious if you have any suggestions.

zfields, 2016-11-13:
I use GoogleTest and GoogleMock on my project. However, they do require you to have GoogleTest installed (easy on Linux via apt install) and a Makefile to run the test. Once they are setup and running, it's quite easy to work with.

Does that sound interesting or like too many dependencies?

soundanalogous, 2016-11-13:
I've been looking into [Catch](https://github.com/philsquared/Catch). Are you familiar with this lib at all? It's just a single header file so super simple to include and runs cross platform.

zfields, 2016-11-14:
I just read the README and it sounds cool. I particularly like the fact that it doesn't need to be installed, and I want to give it a try. I'm in the middle of a prototype, but as soon as I have it complete, I'll throw down on some test and tell you if where if fell short.