firmata/arduino #339: Error compiling :Firmata Arduino

tintinmonster, 2016-12-31:
HI guys, appreciate if anyone could help solve the issue . I ve been trying to link Arduino to processing with Standard firmata example but I kept getting error codes running on windows . Mac is fine but it refuse hangs my processing code.

thanks!!

Arduino: 1.7.11 (Windows 8.1), Board: "Arduino Uno"

In file included from C:\Users\Christine\Desktop\Arduino\libraries\Firmata\utility\SerialFirmata.cpp:20:0:

C:\Users\Christine\Desktop\Arduino\libraries\Firmata\utility\SerialFirmata.h:30:28: fatal error: SoftwareSerial.h: No such file or directory

 #include <SoftwareSerial.h>

                            ^

compilation terminated.

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

soundanalogous, 2016-12-31:
Arduino 1.7.11 is from Arduino.org, not Arduino.cc (there were 2 separate Arduino companies for a while). The Arduino 1.7.x series had many issues in my experience. I recommend upgrading to [Arduino 1.8](https://www.arduino.cc/en/Main/Software) (the 2 companies have now reunited and this is the first joined release). If after upgrading you still see this error let me know. If not, then you can close this issue.

soundanalogous, 2017-01-28:
reopen if this is still an issue

closed 2017-01-28 by soundanalogous