firmata/arduino #342: LEDs inverted with Android Accessory Developer Kit

appsforartists, 2017-01-09:
Using the [ADK Mega + demo shield](https://developer.android.com/adk/adk.html), all three LEDs default to "on," at full brightness, and all values are inverted:

Using Johnny-Five:

```javascript
const led1 = new five.Led.RGB({ pins: [2, 4, 3]});
const led2 = new five.Led.RGB({ pins: [5, 7, 6]});
const led3 = new five.Led.RGB({ pins: [8, 10, 9]});

// This turns the LEDs off
led1.on();
led2.on();
led3.on();

// led2 is the brightest, even though it's set to 1% intensity
led1.intensity(99);
led2.intensity(1);
led3.intensity(99);
```

soundanalogous, 2017-01-09:
The ADK mega demo shield is likely using common anode RGB LEDs.

Try this to inform J5 that the LEDs are common anode rather than the default common cathode configuration:

```js
const led1 = new five.Led.RGB({ pins: [2, 4, 3], isAnode: true});
const led2 = new five.Led.RGB({ pins: [5, 7, 6], isAnode: true});
const led3 = new five.Led.RGB({ pins: [8, 10, 9], isAnode: true});
```

soundanalogous, 2017-01-09:
BTW... I never even thought of using the ADK demo shield with J5. I have one in my desk at work... will have to dust it off now :)

appsforartists, 2017-01-09:
Thanks.  That fixed it.  (I had presumed that since the lights came on from the firmware that the issue was in the firmware.)

Here's what I have so far:

https://gist.github.com/appsforartists/0df4f27ac8f37668813bd2a70e2b18ee

closed 2017-01-09 by appsforartists