firmata/arduino #343: Is this event driven? Does it support fading two lights at the same time?

lostinblue, 2017-01-09:
I have been working on a nearly identical library off and on for two years. The one I have been working on is event driven so it supports very smooth fading between two values.

Your lib is clearly way more supported, so I'm curious if it supports like changing lights at the same time (at least to the human eye), and the ability to print output while taking input using timers. 

Thanks, pardon my ignorance.

soundanalogous, 2017-01-09:
I'm not sure exactly what you are asking:

Here is the [protocol](https://github.com/firmata/protocol) that is implemented by the Arduino Firmata library.

It's all based on loops and callbacks. Timers are not used.

zfields, 2017-02-24:
@lostinblue I'm not sure if this answers your question, but Firmata is the backbone of [this Windows Remote Arduino demo](https://www.youtube.com/watch?v=e-166MsVaAA). In order to paint the LED curtain the phone has to modify 6912 (48*48*3) analog LEDs. I'm not certain how you are trying signal your LEDs but Firmata is extremely capable if you are approaching the problem in a way that lends itself to the protocol.

closed 2017-02-24 by soundanalogous

soundanalogous, 2017-02-24:
@zfields nice demo!