firmata/arduino #346: "stream was not declared in this scope" FirmataStandradBLE

MumfordWiz, 2017-02-09:
Hey,
I'm trying to run the StandardFirmataBLE.
I open the example and compile it, but I get the error:
"'stream' was not declared in this scope" lines: 764, 794
I tried the 2.53 release and the 2.5.4 release, no changes.
I'm using Arduino 1.8.1. Genuino/Uno
Thanks!

soundanalogous, 2017-02-10:
Do you have a compatible BLE shield? If so you need to follow [these instructions in bleConfig.h](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataBLE/bleConfig.h#L22-L48). If not, then you can't run StandardFirmataBLE on an Uno since it does not have BLE.

closed 2017-02-24 by soundanalogous

esridhar126, 2018-02-12:
Hi everyone, I am working on the BLE 4.0, HM-10(CC2540/CC2541) module for my product updation from Bluetooth 2.0. Main controller i am using is Arduino Uno/ Mega. For 2.0 I easily got Firmata from Arduino examples, but BLE4.0 is more advanced one having many specs to interface, i cant get a Firmata from anywhere. In arduino also its available for minimum modules that doesnot  include Arduino mega/Uno.

Is anyone from here suggest me or help how to interface my Microcontroller with API/ APk apps with or without using Standard Firmata BLE. (If it is with standard Firmata it is more helpful.)
My main motive is to control my product with IOS.