firmata/arduino #347: Debouncing digital input pins

troywweber7, 2017-02-14:
It's very common to "debounce" via algorithm the inputs of digital pins to avoid false-positives. Can anyone familiar with this code base inform me as to whether any debouncing is used for digital inputs? I'm assuming I would look within digitalInputs.cpp/.h for this?

soundanalogous, 2017-02-15:
There is no debouncing performed here since a user may not want to debounce a digital input for all cases. Instead debouncing should be performed in the client application. You can see an example [here](https://github.com/soundanalogous/Breakout/blob/master/src/io/Button.js#L84-L109).

closed 2017-02-15 by soundanalogous

troywweber7, 2017-02-15:
I assume this decision is implemented for Configurable Firmata, then, as well. Is there any discussion out there about adding configurable debouncing-timeouts for digital input pins? I.e. as a feature request, perhaps when configuring an input, the user can specify a debounce-timeout that is applied Arduino-side to prevent serial-comm strain?

Obviously measures should be taken mechanically to alleviate debounce (pull-resistors, shunt capacitors, etc). And it also does not hurt to have noise-safeguards client-side. However, the reason I'm looking into this is that I have external pull-up resistors, shunt capacitors, and I still see peculiar behavior. For example, I'm monitoring 3 inputs connected to push-buttons on pins 2, 3, and 4. All are pulled high. When I toggle pin 2, the callback runs not just for 2, but for pins 3 and 4 as well. Interestingly, pin 2 of course shows the proper value (low), but pin 3 and 4 show high. To me this means the inputs went low and then high again before the Arduino's internal loop got around to sending the message.

Now, client-side, this obviously reports the proper value and these erroneous signals can simply be ignored with clever programming JS-side. However, it doesn't quite feel right to me to receive such erroneous signals. Is this something anyone else has run into?

Just as an example, here is a test js-script: https://bitbucket.org/troywweber/firmataexperiments/src/28078d412eb5a8352cf7fa2161a45e08aefec97e/experiments/input-noise/input.js?at=master&fileviewer=file-view-default

Here is the setup I have:

[image]
Please note that while I am using internal pullups, I also added external pullup resistors not pictured here. Resistors added were 330 ohms, caps added are 1 nF.

And when I run it (`node input.js /dev/cu.usbmodemXXXX`) I get these results:

```
$ node input.js /dev/cu.usbmodemFD1121 
Connected StandardFirmata.ino 2.5
setting pin 2 to 11
monitoring pin 2
setting pin 3 to 11
monitoring pin 3
setting pin 4 to 11
monitoring pin 4
pin 2 is HIGH
pin 3 is LOW
pin 4 is LOW
pin 2 is HIGH
pin 3 is HIGH
pin 4 is LOW
pin 2 is HIGH
pin 3 is HIGH
pin 4 is LOW
pin 2 is HIGH
pin 3 is HIGH
pin 4 is HIGH
pin 2 is HIGH
pin 3 is HIGH
pin 4 is HIGH
========== 5s BREAK ==========
### <== here I press button 4
pin 2 is HIGH
pin 3 is HIGH
pin 4 is LOW
### <== here I release button 4
pin 2 is HIGH
pin 3 is HIGH
pin 4 is HIGH
========== 5s BREAK ==========
### <== here I press button 3
pin 2 is HIGH
pin 3 is LOW
pin 4 is HIGH
### <== here I release button 3
pin 2 is HIGH
pin 3 is HIGH
pin 4 is HIGH
========== 5s BREAK ==========
### <== here I press button 2
pin 2 is LOW
pin 3 is HIGH
pin 4 is HIGH
### <== here I release button 2
pin 2 is HIGH
pin 3 is HIGH
pin 4 is HIGH
```

soundanalogous, 2017-02-16:
Have you tried other frameworks besides firmata.js. I'm suspicious the issue you are seeing is on the  client side.

Regarding the ability to enable debounce on digital pins. This does not fit with the existing Firmata digital messaging scheme so this would have to be a new digital pin configuration request. I suggest submitting a feature request and an initial proposal.

ad84, 2017-04-05:
Hardware debounce, see http://www.jeremyblum.com/2011/03/07/arduino-tutorial-10-interrupts-and-hardware-debouncing/
That should work, can pick the response time with the RC circuit.