firmata/arduino #351: Problem connecting ESP8266 (wemos D1 by serial)

olmerg, 2017-02-27:
Hi,
I 'am trying to connect the wemos D1 mini to [snap4arduino](http://snap4arduino.org/), but it only support serial, so I try to connect without success.
Today, I solve the problem, changing the standardFirmdataWifi to Serial Stream and  adding  ignore pins the two pins of the serial port.

```
#if defined(ESP8266_WIFI)
#define IS_IGNORE_PIN(p)  ((p) == 1 || (p) == 3)
```

I do not how to integrate this in the code of the library to work by default.

Thanks

soundanalogous, 2017-02-27:
What I've been trying to avoid, but will ultimately have to do at some point is adding a config.h file to  StandardFirmata and StandardFirmataPlus.

soundanalogous, 2017-02-28:
In the mean time, add the following to StandardFirmata.ino or StandardFirmataPlus.ino (whichever you typically use) just above `Firmata.begin(57600);`

```c++
#ifdef ESP8266
  Firmata.setPinMode(1, PIN_MODE_IGNORE);
  Firmata.setPinMode(3, PIN_MODE_IGNORE);
#endif
```

There is no need to modify StandardFirmataWiFi to accomplish this.

olmerg, 2017-03-01:
Nice, thanks by your soon support.
Yes, after I understood the code I view that is not necessary to work with  StandardFirmataWiFi.

May be a config.h is not a bad idea if there is possible to add new sensors like ultrasound or LCD, just and idea, taking the functions like [firmmata collection](https://github.com/mdcanham/Snap4Arduino-Firmata)

soundanalogous, 2017-03-02:
@olmerg you should look at [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata) which is a module-based approach to Firmata firmware.

Timopheym, 2017-03-03:
Hi! @olmerg, i am trying to connect NodeMCUv3 based on ESP-12E to Snap4Arduino, and it does freely connect j5, BUT on "Connect Arduino"  
And it prints "Could not talk to arduino in port /dev/ttyUSB0" 
[image]
I did tried your change 
```
#if defined(ESP8266_WIFI)
#define IS_IGNORE_PIN(p)  ((p) == 1 || (p) == 3)
```
With StandartFirmataWIFI but it didn't help. 

What am i doing wrong?

Thanks!

Here is it's pins: 
[image]

soundanalogous, 2017-03-03:
Use this sketch: https://gist.github.com/soundanalogous/87ea10c07c3f072cb46d8cad206b42d1

closed 2017-05-21 by soundanalogous

amindadgar, 2021-05-17:
> Use this sketch: https://gist.github.com/soundanalogous/87ea10c07c3f072cb46d8cad206b42d1

Thanks for your support soundanalogous. I tried your code and a code from another person I've attached below, but I still have no success. My board is Arduino Wemos D1 R1 and the board I'm trying to connect it is raspberry pie and the software is node-red.

 
https://github.com/kotl/esp8266-firmata/blob/master/examples/ESP8266FirmataWifi/ESP8266FirmataWifi.ino

amindadgar, 2021-05-19:
> In the mean time, add the following to StandardFirmata.ino or StandardFirmataPlus.ino (whichever you typically use) just above `Firmata.begin(57600);`
>
> ```c++
> #ifdef ESP8266
>   Firmata.setPinMode(1, [...]

After Adding the codes you gave to StandardFirmataPlus.ino ( Or StandardFirmata.ino), I got this error below shown in the picture

[image]

Any help would be really appreciated!

soundanalogous, 2021-05-30:
All of those issues are coming from node-red, not Firmata.