firmata/arduino #355: [v3.0.0] Protocol callback should be a multi-byte command

zfields, 2017-03-12:
It is difficult, if not impossible, to pick up the version number coming across the wire.

soundanalogous, 2017-03-12:
Regarding 0xF9? I'm not sure what the issue is here.

zfields, 2017-03-12:
Sorry for the ultra-terse description - it was 5am (a little past my bedtime ;-)).

The parser callback only provides a signal when it sees the `0xF9`, not the version info. Upon receiving `0xF9` the parser immediately fires the callback (this behavior stems from the fact that the parser used to be exclusively for the Firmata class running on hardware). After receiving `0xF9`, the Arduino turns around and sends '0xF9' followed by the major and minor version (which I can see on the line, but not through the parser). This scenario is fine for the host/server, but it leaves much to be desired on the client side.

In summary, the protocol callback should provide both major and minor version to the callee, and it should be subsequently discarded by the Arduino.

martin-eden, 2019-05-31:
Problem is that version request is (0xF9). And version reply is (0xF9 0x.. 0x..). 

So unsophisticated parser may think that (0xF9 ...) is version request and throw remained bytes.

I'd just use another signature for version reply.