firmata/arduino #362: Add Arduino Primo support for StandardFirmataWiFi

soundanalogous, 2017-03-16:
The Primo has an ESP8266 on-board which works with StandardFirmataWiFi on it's own, but it's connected to the nRF52832 via UART so I'm not sure how that could be translated into how StandardFirmataWiFi currently handles TCP/IP.

soundanalogous, 2017-03-16:
I assume the new [Arduino Uno WiFi board](http://www.arduino.org/products/boards/arduino-uno-wifi) will work in the exact same way.

dapug, 2017-03-28:
Firmata for Uno sends through rx/tx TTL to USB, so I was thinking... why can't an Uno be set up standard like that, and then ESP8266 be connected to serial to "pass through" serial to the network?  

Anyway, this is exactly what I am looking for, but rather than the Uno Wifi board, I want to make use of my Uno and ESP-01 modules I already have plenty of.

soundanalogous, 2017-03-28:
There is a new Arduino Uno Wi-Fi board that is exactly that all in a single board.

dapug, 2017-03-28:
yes, but please see my earlier point... I have existing modules, not interested in getting the all-in-one board.

soundanalogous, 2017-03-28:
Are you interested in putting together an initial PR for this?

soundanalogous, 2017-03-28:
If someone writes firmware for the ESP8266 to act as a UART to Wi-Fi bridge, integrating Firmata support should be fairly trivial.

dapug, 2017-03-28:
I'm not qualified, but would otherwise love to whip up a PR. I can envision what might be entailed, but my code experience is in app building, not firmware/drivers.  I can dig around, maybe someone has done a UART/Wifi bridge for another purpose and we could start there.

soundanalogous, 2017-03-28:
Yeah if you could look around to see if there is an existing UART/WiFi bridge for the ESP, that would be helpful.

dapug, 2017-03-28:
This is as simple as it gets, though I'm not sure about perf or bugs:
https://github.com/roboremo/ESP8266-WiFi-UART-Bridge/blob/master/sketch_esp_WiFi_UART_Bridge.ino

Found a similar one here:
https://github.com/NaoNaoMe/ESP8266-WiFi-UART-transparent-bridge/blob/master/WiFiUartTransparentBridge.ino

If running this as a sketch on the ESP8266 isn't low level enough, I also found some actual firmware changes that allow transparent bridging (far more complicated to get into):
https://github.com/jeelabs/esp-link
https://github.com/beckdac/ESP8266-transparent-bridge

soundanalogous, 2017-03-28:
You could try using either of the 1st 2 examples (2nd one looks best) and changing the baud rate to 57600. Then flash StandardFiramta to the Uno (not StandardFirmataWiFi). Connect the esp8266 serial pins to pins 0 and 1 on the Arduino Uno. Theoretically this should work. You'd obviously have to hard-code your credentials in the config.h file, but it's a start at least.

dapug, 2017-03-29:
ok, here is an interesting report.

Some success, kinda surprised me! Here is exactly what I did.

1. Loaded **StandardFimata** on the **Uno**
2. Loaded **WiFiUartTransparentBridge.ino** UART/wifi bridge (link above) on a 8Mbit ESP-01 using Expressif SDK 1.5.4 and the latest Arduino IDE/core, etc.
3. Took note of the **IP address** of the ESP-01 from the **Serial Monitor** after loading this sketch (because I used DHCP, so I need the IP it got)
4. Connected **GND, RX, TX** of the ESP-01 to **GND, TX, RX** of the Uno, respectively  (* NOTE: My Arduino Uno (Buono R3) can switch and run on **3.3v**, I do not recommend connecting ESP-01 to a 5v Uno like this)
5. Installed and launched the [Windows Arduino Remote Experience](https://www.microsoft.com/en-us/store/p/windows-remote-arduino-experience/9nblggh2041m) app from the MS Store.  (This is designed specifically to test Firmata on Arduino Uno from a Windows 10 app - super slick)
6. Connected to Uno Firmata from the Windows App using the IP of the ESP-01 and port # in the sketch (23, or whatever you set it to)

## Boom, I'm in!

**_BUT_**, here are the issues I found so far:

- It doesn't always connect, not sure why.  I often get this error from the Windows app that is trying to reach it:
```
A device connection was established, but the device failed handshaking procedures. Verify that your device is configured with StandardFirmata. Message: Catastrophic failure

Pin configuration not received.
```

- It doesnt get the pin configuration right when it does get it.  See next...
- It will often list analog pins in the digital list
- It sometimes lists only a few of the analog pins in the actual/proper analog pin list, or none at all
- It sometimes shows me like 50 digital pins (erroneous)

But when it does list pins (such as pin 5 for example) **I can actually send commands and it works** (turn on LED, whatever).  **Analog inputs also work** - when the configuration maps it and doesn't flake out. :(

Given that the Windows app works properly with Firmata on this same Arduino via USB, where do you suppose the problem is?  Flaky TCP to UART translations?  I'm not sure where to go from here, but I sure hope this info is helpful in directing next steps.  Super encouraging so far!

soundanalogous, 2017-03-29:
There are a couple of possible causes here:

1. The capability query response which contains the pin info is huge, I think around 500 bytes so there is a good chance due to the relaying of data that a buffer is occasionally getting overrun on the ESP8266 side. One solution here is for the client library to provide a way to skip the capability query and allow the user to supply a pin map in its place (firmata.js and johnny-five provide this).

2. Because it takes some time to connect to the AP and Firmata client's may depend on an initial ping from the board, depending on the startup sequence the particular client uses it may be missed from time-to-time.

dapug, 2017-03-29:
Regarding this code in the sketch:

```
  //check a client for data
  if (serverClient && serverClient.connected()){
    if(serverClient.available()){
      size_t len = serverClient.available();
      uint8_t sbuf[len];
      serverClient.readBytes(sbuf, len);
      Serial.write(sbuf, len);      
    }
  }
```

Is it trying to hold out for the entire 500 bytes on readBytes(), using a 256 byte buffer?  If that is the case, yeah, I can see how that might be a problem.  But can't this be resolved by modifying the read/write sequence here to make sure buffer length is not exceeded?

Back to the original topic of this thread... I assume these kinds of issues may need to be addressed for Primo and the Uno Wifi board.  If I'm wrong about that, sorry, I should start a new thread!

soundanalogous, 2017-03-29:
Based on what I've seen for the Primo and Uno WiFi, they appear to use a version of the WiFi library so they would likely use StandardFirmataWiFi rather than StandardFirmata.

I recommend opening a separate issue for the ESP UART <-> Wi-Fi bridge.

gokhandilek, 2017-04-02:
I also wonder if StandardFirmataWifi would work on Arduino Yun.
Has anybody tried with UnoWifi? Can you connect your firmata app running on another host to arduino via wifi?

soundanalogous, 2017-04-02:
StandardFirmataWiFi does not work on the Arduino Yun, but StandardFirmataEthernet does.

gokhandilek, 2017-04-02:
Thanks for that. 
I am planning to but arduino uno wifi. Have you tried it using arduino uno wifi?

soundanalogous, 2017-04-03:
I do not have Arduino Uno WiFi. I'm not sure what it will take to get StandardFirmataWiFi to work with the Uno WiFi since I'm not sure how the communication is setup between the Uno and the ESP8266 on the board.

soundanalogous, 2017-04-03:
If you want to use WiFi with Firmata I recommend an Arduino MKR1000 or an Adafruit Huzzah ESP8266.

gokhandilek, 2017-04-03:
The only problem with the MKR1000 is that it operates at 3.3V and lower amperage compared to Arduino Uno .
I have a stepper driver which expects the digital inputs to be at higher than 3.5V.
Do you think I can get away with this ?
[image]
Otherwise, I will give mkr1000 a try to get the firmatawifi.