firmata/arduino #363: Firmata strings encoded incorrectly

zfields, 2017-03-18:
This needs a rev. I realized what I did when I was messing around today. I had encoded Firmata strings directly (including the Firmware name), but I couldn't detect it because I had parsed the firmware incorrectly.

I was refactoring the parser's firmware string decoder to use your logic for strings, when I realized what I had done.

I have the pull request ready to patch...

zfields, 2017-03-18:
I discovered it on the ESP8266, because your node client parses the firmware name. I didn't see it before, because the analog data scrolls so fast and I never scrolled back up to the top.

soundanalogous, 2017-03-18:
I2C also appears to be broken in 2.5.5

soundanalogous, 2017-03-18:
Your latest update fixes the I2C issue as well. Looks like in 2.5.5 all sysex data was being parsed as strings.

closed 2017-03-18 by soundanalogous