firmata/arduino #372: Serial Port Dependency

Qaoud, 2017-06-27:
Hi Julian!

I have a problem with Firmata.js which is "Serialport dependency" i don't know what does it mean!! 
and i want to know if that code is right or not:

`/**
 * @author Julian Gautier
 */

/**
 * Module Dependencies
 */

var util = require('util'),
    events = require('events'),
    chrome = chrome || undefined,
    Encoder7Bit = require('./encoder7bit'),
    OneWireUtils = require('./onewireutils'),
    SerialPort = null;

try {
    if (typeof chrome !== "undefined" && chrome.serial) {
        SerialPort = require('browser-serialport').SerialPort;
    } else {
        SerialPort = require('serialport').SerialPort;
    }
} catch (err) {
    SerialPort = null;
}

if (SerialPort == null) {
    console.log("It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install");
    throw "Missing serialport dependency";
}

/**
 * constants
 */

var PIN_MODE = 0xF4,
    REPORT_DIGITAL = 0xD0,
    REPORT_ANALOG = 0xC0,
    DIGITAL_MESSAGE = 0x90,
    START_SYSEX = 0xF0,
    END_SYSEX = 0xF7,
    QUERY_FIRMWARE = 0x79,
    REPORT_VERSION = 0xF9,
    ANALOG_MESSAGE = 0xE0,
    CAPABILITY_QUERY = 0x6B,
    CAPABILITY_RESPONSE = 0x6C,
    PIN_STATE_QUERY = 0x6D,
    PIN_STATE_RESPONSE = 0x6E,
    ANALOG_MAPPING_QUERY = 0x69,
    ANALOG_MAPPING_RESPONSE = 0x6A,
    I2C_REQUEST = 0x76,
    I2C_REPLY = 0x77,
    I2C_CONFIG = 0x78,
    STRING_DATA = 0x71,
    SYSTEM_RESET = 0xFF,
    PULSE_OUT = 0x73,
    PULSE_IN = 0x74,
    SAMPLING_INTERVAL = 0x7A,
    STEPPER = 0x72,
    ONEWIRE_DATA = 0x73,

    ONEWIRE_CONFIG_REQUEST = 0x41,
    ONEWIRE_SEARCH_REQUEST = 0x40,
    ONEWIRE_SEARCH_REPLY = 0x42,
    ONEWIRE_SEARCH_ALARMS_REQUEST = 0x44,
    ONEWIRE_SEARCH_ALARMS_REPLY = 0x45,
[... 936 lines cut ...]
            accel & 0x7F,
            (accel >> 7) & 0x7F,
            decel & 0x7F,
            (decel >> 7) & 0x7F
        );
    }
    data.push(END_SYSEX);
    this.sp.write(new Buffer(data));
    this.once('stepper-done-'+deviceNum, callback);
};

/**
 * Send SYSTEM_RESET to arduino
 */

Board.prototype.reset = function () {
    this.sp.write(new Buffer([SYSTEM_RESET]));
};

module.exports = {
    Board: Board,
    SYSEX_RESPONSE: SYSEX_RESPONSE,
    MIDI_RESPONSE: MIDI_RESPONSE
};
`

rwaldron, 2017-06-27:
When you install this module via npm, npm will also install firmata.js's dependencies, which includes node-serialport. The code you've pasted is the entire source of firmata.js module, and I'm not sure why? Can you paste the code you have that's problematic?

Qaoud, 2017-06-27:
i have a problem with serialport. i'm doing a project which is a robotic hand i can control it with my hand using LeapMotion controller and Arduino Uno and i'm using Node.js to the LeapMotion communicate with the Adruino. I have everything installed, but when i try to run the app that written in javascript a message appears saying issue throw serialport dependencies. I'll upload the files that i'm using to let you see it, and i hope you can help me. 
[F29DBFRHUPHK4ZJ (1).zip](https://github.com/firmata/arduino/files/1106490/F29DBFRHUPHK4ZJ.1.zip)

soundanalogous, 2017-06-28:
@Qaoud did you install the StandardFirmata firmware on your Arduino?

Also, no one is going to download and open a zip posted to a public forum, please post your code to a Github gist.

Qaoud, 2017-06-28:
yes i did and i don't really know where's the problem because i tried other project also using arduino and leapmotion and it works, but with this one the one i need because it's my graduation project doesn't work!! and i don't know if the serial port is working or not!! this's the message i got when i run the program in Node.js:
`It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install

/Users/M.Qaoud/Desktop/Tesi/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:28
    throw "Missing serialport dependency";
    ^
Missing serialport dependency`

soundanalogous, 2017-06-28:
Try the following:

```
npm uninstall firmata
npm install serialport
npm install firmata
```

Qaoud, 2017-06-28:
i did this but the same problem (serialport dependencies)!!

Qaoud, 2017-06-28:
https://www.instructables.com/id/Robotic-Hand-controlled-by-Gesture-with-Arduino-Le/
this's the project that i'm trying  to do! @soundanalogous

soundanalogous, 2017-06-28:
This forum is for issues with the Firmata firmware that you install on an Arduino board, but the issue you are experiencing is with the node serialport module. Please browse the issues in this repository and file an issue there if you don't find anything that helps you to resolve it: https://github.com/EmergingTechnologyAdvisors/node-serialport.

closed 2017-06-28 by soundanalogous

soundanalogous, 2017-06-28:
Also try the node-serialport gitter channel: https://gitter.im/EmergingTechnologyAdvisors/node-serialport