firmata/arduino #384: Firmata for esplora

DubbyLabby, 2017-09-19:
Hi,
Can someone point me in the right direction to understand if it's possible use booth?
It will require some adaptation to the code, etc?

Thanks.
Db

soundanalogous, 2017-09-19:
It should just work. Have you tried uploading StandardFirmata to an Esplora? You can see which pins are mapped to which components here: https://github.com/arduino-libraries/Esplora/blob/master/src/Esplora.cpp.

DubbyLabby, 2017-09-19:
Hi Jeff!
I still not since I was using arduino duemillanove back in the days and I'm coming back to tinkering after a 5 years pause... I'm going to give a try to the esplora after your answer by sure! Back in the days it was a bit more complex to code arduino for non-coders like me. Now I see this a great tool for learning and even on kids <3

So if I understand well firmata should work as is without any library adjustment? Since esplora has its own library and dedicated io I was a bit confused.
I will give a try!

Thanks a lot!
David

soundanalogous, 2017-09-21:
I think it should work fine. The Esplora is really just a Leonardo (ATMega32u4 microcontroller) with a bunch inputs and outputs built into the board. These inputs and outputs are wired to the normal digital and analog pins on the ATMega32u4 so it shouldn't be any different than using a Leonardo with those same inputs and outputs attached separately. The joystick may be kind of tricky. You'd have to dig into the implementation for the johnny-five Joystick example for the esplora: http://johnny-five.io/examples/joystick-esplora/.

DubbyLabby, 2017-09-21:
Awesome! 

Thanks a lot,
David

Dubby at the Labby.

closed 2018-01-01 by soundanalogous