firmata/arduino #386: From one day to the next, Arduino doesn't work properly

weezixx, 2017-12-13:
Hi ! (English is not my mother tong, so I'm already sorry)

Short version:

I use an Arduino Mega, with Pyfirmata.

Yesterday I use this function : 

def moteur_capteur(moteur, capteur):
    board.digital[moteur].write(1)

    while True:
        if capteur.read() == False:
            board.digital[moteur].write(0)
            time.sleep(1)
            break

So it turn ON the motor, and when the sensor is touch (False), the motor stop.

Yesterday, it worked perfectly, and today … well not any more.

And I totally don't get why !

Thanks for your help.

soundanalogous, 2017-12-13:
Can you paste in your full script or link to it in a gist?

weezixx, 2018-01-03:
Sorry I'm late !
I found a solution, without solving the problem ^^
I just create a copy of the file (with an other name), and when the problem occur I switch with the other file … And everything work perfectly.

Thanks again, and … see ya ;)

naiveai, 2018-08-28:
Please close the issue, as it is solved.

closed 2018-08-31 by soundanalogous