firmata/arduino #387: Firmata via SoftwareSerial not works

VashTheProgrammer, 2017-12-18:
Hi, 

if I use this code on StandardFirmataPlus, the firmata works on the begin but then it stop when I send Firmata.digitalwrite from my client program.

Code into StandardFirmataPlus:

void setup()
{
  Serial.begin(115200); // for debugging
  Serial.println("644p starting ...");
  
  ...

  SoftwareSerial  VirtualSerial(2, 4); // (RX, TX)
  VirtualSerial.begin(9600);

  Firmata.begin(VirtualSerial);

  systemResetCallback();  // reset to default config
}

In my client: 
(I use FirmataC)

#include "firmata.h"

int             main()
{
  t_firmata     *firmata;
  int           i = 0;

  firmata = firmata_new("/dev/ttyUSB0"); //init Firmata

  while(!firmata->isReady) //Wait until device is up
    firmata_pull(firmata);

  firmata_pinMode(firmata, 13, MODE_OUTPUT); //set pin 13 (led on most arduino) to out
  while (1)
    {
      sleep(2);
      if (i++ % 2)
        firmata_digitalWrite(firmata, 13, HIGH); //light led
      else
        firmata_digitalWrite(firmata, 13, LOW); //unlight led
    }
}
`

The same code but with an Hardware Serial works.

soundanalogous, 2017-12-18:
Which board are you using?

VashTheProgrammer, 2017-12-18:
Arduino uno and last ide arduino.cc

soundanalogous, 2017-12-18:
Oh, you are trying to use Firmata over SoftwareSerial instead of using SoftwareSerial as a Firmata feature. That is not supported. You have to use hardware serial for both boards.

VashTheProgrammer, 2017-12-18:
Yes I understand that it is no supported but I don t understand why. I
think because firmata use full duplex communication and software serial not
support it but I think to give a support for software serial is not
complicated ... Why don t you spend time for it ?  I tried to use an
additional gpio for sync and disabled some functions and it works but it is
a workaround. Can we work to find a better solution ? What do you think
about it ?

soundanalogous, 2017-12-19:
Try [AltSoftSerial](https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html) instead of the default SoftwareSerial library.

soundanalogous, 2017-12-19:
I also do not recommend using StandardFirmataPlus for this, use StandardFirmata instead. The way StandardFirmataPlus currently uses SoftwareSerial is for interfacing with serial devices such as GPS receivers, rather as a Firmata transport as you are trying to use it in your code example above. This may actually be causing a conflict.

VashTheProgrammer, 2017-12-28:
Sorry, but it doesn't works.

Do you think to support normal SoftwareSerial library for Firmata?

Is it in your to do list ?

Thanks

soundanalogous, 2017-12-28:
It's not on my todo list. What is your use case?

VashTheProgrammer, 2017-12-28:
Esp32 talks with a avr microcontrollor when run firmata. In the esp32 i use
a firmata client and hardware serial. Instead in avr I use software serial
and firmata standard sketch.    I have another wire for communication
between avr and esp32.

soundanalogous, 2017-12-28:
Are you using an ATTiny (or other MCU without a UART) for the avr and that is why you need to use software serial?

VashTheProgrammer, 2017-12-28:
I have a complete board, and I can't change the route a this release
The name is Fluo WiFi.

www.fluotechnology.co.uk

soundanalogous, 2017-12-28:
I need to understand exactly what you are trying to accomplish and why software serial is your only option.

VashTheProgrammer, 2017-12-28:
Don't you have a Firmata light version that probably it works on
softwareSerial ?

I need to control only DigitalWrite/DigitalRead and AnalogWrite/AnalogRead
at the moment and maybe SERVO.

soundanalogous, 2017-12-29:
Go to [firmatabuilder.com](http://www.firmatabuilder.com) and select just those 4 features (the top 4 'Core features'). Leave out Servo for now since both Servo and SoftwareSerial use some of the same interrupts and may cause a conflict depending on which pins you use for servos vs SW serial TX/RX.

After you've downloaded the sketch from FirmataBuilder, replace `Firmata.begin(57600)` with the following lines:

```c++
SoftwareSerial VirtualSerial(2, 4); // (RX, TX)
VirtualSerial.begin(9600); // a higher baud rate may work better, but don't exceed 57600 baud.
// This PIN_MODE_IGNORE step is important or when systemResetCallback() is called,
// the serial RX pin will be changed to an OUTPUT pin.
Firmata.setPinMode(2, PIN_MODE_IGNORE);
Firmata.setPinMode(4, PIN_MODE_IGNORE);

Firmata.begin(VirtualSerial);
```

Be sure to have the [ConfigurableFirmata library](https://github.com/firmata/ConfigurableFirmata) installed.

soundanalogous, 2017-12-29:
Also make sure you don't use PWM or any digital I/O on the same AVR port as the SW serial RX/TX pins.