firmata/arduino #392: Issue with StandardFirmataWIFI using AP mode

jguille2, 2018-02-24:
Hi!
I want to use Firmata with the ESP8266.

I tried StandardFirmataWiFi and it's ok (Thanks!!!) It has different options... but  it connects to WiFi always in _Station mode_ (connecting to a pre-existing WiFi network).

Then, I've tried to add to StandardFirmataWiFi another option to use _AP mode_... but I 've had a problem.

I see this problem is with the `WiFi.status()` command. It is working in _AP mode_ in a different way.

I will report this to the ESP8266/Arduino project (I think current behavior is not right)... but in the meantime, we can change [maintain function in WiFiServerStream.h](https://github.com/firmata/arduino/blob/master/utility/WiFiServerStream.h#L78), changing
`if ( !_listening && WiFi.status() == WL_CONNECTED )`
for
`if ( !_listening && ( WiFi.status() == WL_CONNECTED || WiFi.getMode() == WIFI_AP ))`

and then,we will be able to implement _stream_ sketches in this _AP mode_.

If you think this change is right... I will try to implement this _AccessPoint option_ into StandardFirmataWiFi.ino example.

Thanks for all (I enjoy using firmata!!)

Joan

soundanalogous, 2018-02-25:
If `WiFi.getMode()` and `WIFI_AP` are only defined for ESP8266, then that statement would have to be specific to ESP8266. Otherwise I don't see any issue in making that change.

jguille2, 2018-02-25:
Thanks Jeff,

I have already reported this problem to esp8266/arduino... but according to your words, I'll send a PR with that change, doing it specific to ESP8266. It will be:
```javascript
#ifdef ESP8266
if ( !_listening && ( WiFi.status() == WL_CONNECTED || WiFi.getMode() == WIFI_AP ))
#else
if ( !_listening && WiFi.status() == WL_CONNECTED )
#endif
{
// start TCP server after first WiFi connect
...
```
Is it fine? Thanks!

Joan

soundanalogous, 2018-02-26:
I've been trying to avoid adding any #ifdefs in WiFiServerStream.h. I need to think about this a bit.