firmata/arduino #398: Avoid including Serial library when not needed

mircobabini, 2018-05-24:
There are some libraries out there that conflicts with Serial.
An example overall: Conceptinetics DMX library. check this up https://forum.arduino.cc/index.php?topic=523418.0

By the way I was able to make it work thanks to firmata + eth!
Commenting out these lines everything compiles as expected.

`void FirmataClass::begin(long speed)
{
  Serial.begin(speed);
  blinkVersion();
  begin(Serial);
}`

Please avoid including Serial even when not needed.
These rows are (almost) uses if you use StandardFirmataEthernet w/o debug features.

soundanalogous, 2018-05-25:
This can't be done in a backwards compatible way. It would break all projects that depend on the current version of Firmata. I can consider it for Firmata 3.0.

mircobabini, 2018-05-25:
First of all: thank you. It would be great.

I also checked the configurable firmata but I think that it’s affected by the same thing. I wasn’t able to find a Firmata release that totally doesn’t use the Serial lib.

Well, thanks you again.