firmata/arduino #399: VB Firmata control ESP-01 relay

brian12112, 2018-05-24:
Thanks for this great library. I has done a project using it. Now I have plan to do it with esp8266 so I have a questions / need help,

I wonder if there is a example how to send "SimpleDigitalFirmata" to control relay? Im using Arduino Uno with Ethernet shield and 3x ESP-01.  

My planning is to read VB firmata from pc using Uno and send to ESP using Ethernet Shield.

Thank You...

brian12112, 2018-05-30:
Hello Mr. Jeff Hoefs/soundanalogous, I hope you not busy. I need your help for my problem above, or maybe you have suggestion what should i do? Thank You

soundanalogous, 2018-06-01:
You can use the ESP8266 with WiFi without the need for an Arduino Uno in the middle. Use StandardFirmataWiFi. See the notes in [wifiConfig.h](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataWiFi/wifiConfig.h) for using with an ESP8266 board.

brian12112, 2018-06-05:
Thank you Mr. Jeff, Its a great start for me. Hope its not to hard, because I dont know not much about esp.

closed 2018-06-05 by brian12112